realsenseai/librealsense #6966: Synchronization of multiple LiDAR cameras

Kepler219b, 2020-07-29:
I am looking for a possibility to synchronize a LiDAR L515 camera with another LiDAR or a D435 camera. But how can I do this? I have read, that the hardware sync connection of the LiDAR camera is similar to the D435. But where can I find the correct pin description and port dimensions of that.

Thanks a lot for your help.

dorodnic, 2020-07-29:
A white-paper on this subject will be published soon

zapaishchykova, 2020-07-29:
Also interested!

RealSenseSupport, 2020-08-04:
Thanks for the interest, we are working to get the document released as soon as we can.

We have published the following document, while it's not the sync document, it is worth a look:

https://support.intelrealsense.com/hc/en-us/articles/360051646094-Intel-RealSense-LiDAR-Camera-L515-User-Guide

agrunnet, 2020-08-05:
The white paper on this was just published on RealSense website. Hope it is helpful.

closed 2020-08-05 by agrunnet

zapaishchykova, 2020-08-05:
Link?

agrunnet, 2020-08-05:
https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup