realsenseai/librealsense #6972: Android tombstone from invoke_callback

joeyf123, 2020-07-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | ( 05.12.03.00) | 
| Operating System & Version |   { Android 10 }  | 
| Platform                                 | { Android } |
| SDK Version                            |  { 2.36.0 }                          | 
| Language                            |  { JAVA }                          | 
| Segment			|  {Smartphone }                          | 

### Issue Description
Getting a tombstone crash while using the realsense camera on android:

### Code
`        
@Override
        public void run() {

            try (FrameReleaser fr = new FrameReleaser()) {
                FrameSet frames = mPipeline.waitForFrames().releaseWith(fr);
                FrameSet orgSet = frames.releaseWith(fr).applyFilter(mPointCloud).releaseWith(fr);
                FrameSet processedSet = frames.releaseWith(fr)
                        .applyFilter(mDecimationFilter).releaseWith(fr)
                        .applyFilter(mDisparity).releaseWith(fr)
                        .applyFilter(mSpatialFilter).releaseWith(fr)
                        .applyFilter(mTemporalFilter).releaseWith(fr)
                        .applyFilter(mHoleFillingFilter).releaseWith(fr)
                        .applyFilter(mDisparity).releaseWith(fr)
                        .applyFilter(mColorizer).releaseWith(fr);

                Frame colorDepthFrame = processedSet.first(StreamType.DEPTH).releaseWith(fr);
                Frame texture = processedSet.first(StreamType.COLOR, StreamFormat.RGB8).releaseWith(fr);
                DepthFrame depth = orgSet.first(StreamType.DEPTH, StreamFormat.Z16).releaseWith(fr).as(Extension.DEPTH_FRAME);
                Frame pointCloudFrame = orgSet.first(StreamType.DEPTH, StreamFormat.XYZ32F).releaseWith(fr);`

### Exception:

2020-07-29 08:58:38.025 11517-11517/? A/DEBUG: backtrace:
2020-07-29 08:58:38.025 11517-11517/? A/DEBUG:       #00 pc 00000000007c2f0c  /data/app/com.example.optiflow-smQ9IEnw63HlSN9D6NXJng==/base.apk!librealsense2.so (offset 0x8b8000) (librealsense::frame_source::invoke_callback(librealsense::frame_holder) const+76) (BuildId: 886c1e59416f6e60a49ce78b8aa5c5972cb4a41e)
2020-07-29 08:58:38.025 11517-11517/? A/DEBUG:       #01 pc 0000000000661f2c  /data/app/com.example.optiflow-smQ9IEnw63HlSN9D6NXJng==/base.apk!librealsense2.so (offset 0x8b8000) (librealsense::synthetic_source::frame_ready(librealsense::frame_holder)+48) (BuildId: 886c1e59416f6e60a49ce78b8aa5c5972cb4a41e)
2020-07-29 08:58:38.025 11517-11517/? A/DEBUG:       #02 pc 00000000007692f4  /data/app/com.example.optiflow-smQ9IEnw63HlSN9D6NXJng==/base.apk!librealsense2.so (offset 0x8b8000) (rs2_synthetic_frame_ready+84) (BuildId: 886c1e59416f6e60a49ce78b8aa5c5972cb4a41e)
2020-07-29 08:58:38.025 11517-11517/? A/DEBUG:       #03 pc 00000000006643b4  /data/app/com.example.optiflow-smQ9IEnw63HlSN9D6NXJng==/base.apk!librealsense2.so (offset 0x8b8000) (BuildId: 886c1e59416f6e60a49ce78b8aa5c5972cb4a41e)
2020-07-29 08:58:38.025 11517-11517/? A/DEBUG:       #04 pc 000000000065fa64  /data/app/com.example.optiflow-smQ9IEnw63HlSN9D6NXJng==/base.apk!librealsense2.so (offset 0x8b8000) (librealsense::processing_block::invoke(librealsense::frame_holder)+96) (BuildId: 886c1e59416f6e60a49ce78b8aa5c5972cb4a41e)
2020-07-29 08:58:38.025 11517-11517/? A/DEBUG:       #05 pc 0000000000772da8  /data/app/com.example.optiflow-smQ9IEnw63HlSN9D6NXJng==/base.apk!librealsense2.so (offset 0x8b8000) (rs2_process_frame+76) (BuildId: 886c1e59416f6e60a49ce78b8aa5c5972cb4a41e)
2020-07-29 08:58:38.025 11517-11517/? A/DEBUG:       #06 pc 00000000006e52cc  /data/app/com.example.optiflow-smQ9IEnw63HlSN9D6NXJng==/base.apk!librealsense2.so (offset 0x8b8000) (Java_com_intel_realsense_librealsense_ProcessingBlock_nInvoke+84) (BuildId: 886c1e59416f6e60a49ce78b8aa5c5972cb4a41e)
2020-07-29 08:58:38.025 11517-11517/? A/DEBUG:       #07 pc 00000000000f2e50  /data/app/com.example.optiflow-smQ9IEnw63HlSN9D6NXJng==/oat/arm64/base.odex (art_jni_trampoline+160)
2020-07-29 08:58:38.025 11517-11517/? A/DEBUG:       #08 pc 000000000204d8f4  /memfd:/jit-cache (deleted) (com.intel.realsense.librealsense.Filter.process+84)
2020-07-29 08:58:38.025 11517-11517/? A/DEBUG:       #09 pc 0000000002050184  /memfd:/jit-cache (deleted) (com.intel.realsense.librealsense.FrameSet.applyFilter+68)
2020-07-29 08:58:38.026 11517-11517/? A/DEBUG:       #10 pc 000000000201f920  /memfd:/jit-cache (deleted) (com.example.optiflow.RealSenseVideo$1.run+272)
2020-07-29 08:58:38.026 11517-11517/? A/DEBUG:       #11 pc 000000000203f404  /memfd:/jit-cache (deleted) (android.os.Handler.handleCallback+68)
2020-07-29 08:58:38.026 11517-11517/? A/DEBUG:       #12 pc 0000000002033550  /memfd:/jit-cache (deleted) (android.os.Handler.dispatchMessage+64)
2020-07-29 08:58:38.026 11517-11517/? A/DEBUG:       #13 pc 0000000000136334  /apex/com.android.runtime/lib64/libart.so (art_quick_invoke_stub+548) (BuildId: db23cd677db3df94c9fa4fd1407e7af4)
2020-07-29 08:58:38.026 11517-11517/? A/DEBUG:       #14 pc 0000000000144fec  /apex/com.android.runtime/lib64/libart.so (art::ArtMethod::Invoke(art::Thread*, unsigned int*, unsigned int, art::JValue*, char const*)+244) (BuildId: db23cd677db3df94c9fa4fd1407e7af4)
2020-07-29 08:58:38.026 11517-11517/? A/DEBUG:       #15 pc 00000000002e27fc  /apex/com.android.runtime/lib64/libart.so (art::interpreter::ArtInterpreterToCompiledCodeBridge(art::Thread*, art::ArtMethod*, art::ShadowFrame*, unsigned short, art::JValue*)+384) (BuildId: db23cd677db3df94c9fa4fd1407e7af4)
2020-07-29 08:58:38.026 11517-11517/? A/DEBUG:       #16 pc 00000000002dda5c  /apex/com.android.runtime/lib64/libart.so (bool art::interpreter::DoCall<false, false>(art::ArtMethod*, art::Thread*, art::ShadowFrame&, art::Instruction const*, unsigned short, art::JValue*)+892) (BuildId: db23cd677db3df94c9fa4fd1407e7af4)
2020-07-29 08:58:38.026 11517-11517/? A/DEBUG:       #17 pc 000000000059fcac  /apex/com.android.runtime/lib64/libart.so (MterpInvokeVirtual+648) (BuildId: db23cd677db3df94c9fa4fd1407e7af4)
2020-07-29 08:58:38.026 11517-11517/? A/DEBUG:       #18 pc 0000000000130814  /apex/com.android.runtime/lib64/libart.so (mterp_op_invoke_virtual+20) (BuildId: db23cd677db3df94c9fa4fd1407e7af4)
2020-07-29 08:58:38.026 11517-11517/? A/DEBUG:       #19 pc 00000000003210e2  /system/framework/framework.jar (android.os.Looper.loop+466)
2020-07-29 08:58:38.026 11517-11517/? A/DEBUG:       #20 pc 00000000005a2a1c  /apex/com.android.runtime/lib64/libart.so (MterpInvokeStatic+1040) (BuildId: db23cd677db3df94c9fa4fd1407e7af4)
2020-07-29 08:58:38.026 11517-11517/? A/DEBUG:       #21 pc 0000000000130994  /apex/com.android.runtime/lib64/libart.so (mterp_op_invoke_static+20) (BuildId: db23cd677db3df94c9fa4fd1407e7af4)
2020-07-29 08:58:38.027 11517-11517/? A/DEBUG:       #22 pc 000000000018ca40  /system/framework/framework.jar (android.app.ActivityThread.main+196)
2020-07-29 08:58:38.027 11517-11517/? A/DEBUG:       #23 pc 00000000002b3b10  /apex/com.android.runtime/lib64/libart.so (_ZN3art11interpreterL7ExecuteEPNS_6ThreadERKNS_20CodeItemDataAccessorERNS_11ShadowFrameENS_6JValueEbb.llvm.15427027571776522123+240) (BuildId: db23cd677db3df94c9fa4fd1407e7af4)
2020-07-29 08:58:38.027 11517-11517/? A/DEBUG:       #24 pc 0000000000591214  /apex/com.android.runtime/lib64/libart.so (artQuickToInterpreterBridge+1032) (BuildId: db23cd677db3df94c9fa4fd1407e7af4)
2020-07-29 08:58:38.027 11517-11517/? A/DEBUG:       #25 pc 000000000013f468  /apex/com.android.runtime/lib64/libart.so (art_quick_to_interpreter_bridge+88) (BuildId: db23cd677db3df94c9fa4fd1407e7af4)
2020-07-29 08:58:38.027 11517-11517/? A/DEBUG:       #26 pc 00000000001365b8  /apex/com.android.runtime/lib64/libart.so (art_quick_invoke_static_stub+568) (BuildId: db23cd677db3df94c9fa4fd1407e7af4)
2020-07-29 08:58:38.027 11517-11517/? A/DEBUG:       #27 pc 000000000014500c  /apex/com.android.runtime/lib64/libart.so (art::ArtMethod::Invoke(art::Thread*, unsigned int*, unsigned int, art::JValue*, char const*)+276) (BuildId: db23cd677db3df94c9fa4fd1407e7af4)
2020-07-29 08:58:38.027 11517-11517/? A/DEBUG:       #28 pc 00000000004afc18  /apex/com.android.runtime/lib64/libart.so (art::(anonymous namespace)::InvokeWithArgArray(art::ScopedObjectAccessAlreadyRunnable const&, art::ArtMethod*, a [... cut ...]
2020-07-29 08:58:38.028 11517-11517/? A/DEBUG:       #49 pc 00000000000c282c  /system/lib64/libandroid_runtime.so (android::AndroidRuntime::start(char const*, android::Vector<android::String8> const&, bool)+776) (BuildId: 87400b7ec8d52189a54ca10d6f074004)
2020-07-29 08:58:38.028 11517-11517/? A/DEBUG:       #50 pc 00000000000035b0  /system/bin/app_process64 (main+1376) (BuildId: 1a86d090ff0d3dd5f5a671601a88a247)
2020-07-29 08:58:38.029 11517-11517/? A/DEBUG:       #51 pc 000000000007e898  /apex/com.android.runtime/lib64/bionic/libc.so (__libc_init+108) (BuildId: 91cd22218589ba7b9ea39ce46bc68dff)

Thanks in advance.

MartyG-RealSense, 2020-07-29:
Hi @joeyf123   There have been a couple of other past cases of Android tombstone crashes that resemble your error log.

https://github.com/IntelRealSense/librealsense/issues/5524
https://github.com/IntelRealSense/librealsense/issues/6087

A characteristic of those other cases is that it may not occur every time the program is run, or may not occur until some time after the program has started running.    Your mention of getting a crash "whilst using" Android suggests that you also experience the crash at a time after the application has been running successfully.  Can you confirm whether this is correct, please?

I note that you are using Android 10, which can have conflicts with the RealSense Android wrapper.  Could you try setting **targetSdkVersion** to a value of '27' please (the fix for Android 10 incompatibility) to see whether your problems with tombstoning still occur after that please?

joeyf123, 2020-07-29:
You hit the nail on the head. I am getting the crash consistently under 1 minute of use and it will occur every time. The targetSdkVersion is already set to 27 as you specified.

CompileSDKVersion:       30
BuildToolsVersion:           29.0.3
TargetSDKVersion:          27
MinSDKVersion:               24

#5524<https://github.com/IntelRealSense/librealsense/issues/5524>
#6087<https://github.com/IntelRealSense/librealsense/issues/6087>

MartyG-RealSense, 2020-07-29:
It looks as though you are applying a lot of post-processing filters in your script, and that is likely to have a consequence as post-processing is performed on the device's processor and not in the camera hardware.  It may be worth removing the filters (or at least keeping Decimation) and see how long the application can then run before crashing.

MartyG-RealSense, 2020-08-08:
Hi @joeyf123   Do you require further assistance with this case please, or can it be closed?   Thanks!