realsenseai/librealsense #6976: GLFW Driver Error

ameth64, 2020-07-30:
| | |
|---|---|
|**librealsense**|2.37.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
GLFW Driver Error: WGL: The driver does not appear to support OpenGL
Could not initialize offscreen context!

```

MartyG-RealSense, 2020-07-30:
Hi @ameth64   This is not a problem that is specific to librealsense and the same error can occur in other programs (most commonly the videogame **Minecraft**).  It suggests that **OpenGL** graphics support is not present on your computer.  

Are you using your computer in a "headless" setup (one without a display screen) please?

dorodnic, 2020-08-05:
The error seem fairly descriptive. Please install graphic drivers and try again

closed 2020-08-05 by dorodnic