realsenseai/librealsense #6977: Is there a paper & source code for Deep Learning for VR/AR: Body tracking with Intel RealSense Technology?

michelodu, 2020-07-30:
Hello, 

I have an project in mind similar to the set-up described by the video Deep Learning for VR/AR: Body tracking with Intel RealSense Technology, but I would like to find out more. The only papers by Dr Krejov date back to 2017 and center on his Random Forests approach along with the model-based component, whereas this one clearly mentions a convolutional neural network. Is the code for this available, and is there a paper that I can read that describes the design in detail? 

Best wishes, 

Michel

MartyG-RealSense, 2020-07-30:
Hi @michelodu   The YouTube video that you mention ( https://www.youtube.com/watch?v=VSHDyUXSNqY&t=678s )  was originally a real-world seminar that was later published to YouTube, so it does not have accompanying resources

[image]

If you want to explore use of a Convolutional Neural Network (CNN), a good starting point may be the **OpenVINO Toolkit**, which has a RealSense compatibility wrapper.

**OpenVINO**
https://software.intel.com/content/www/us/en/develop/tools/openvino-toolkit.html

**RealSense OpenVINO wrapper**
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/openvino

michelodu, 2020-07-30:
Thanks, Marty. I'm a bit bummed about the lack of info from Dr Krejov, but I have more info at my disposal now than before you and I began this discussion. Best wishes, Michel

MartyG-RealSense, 2020-08-08:
As this case seems to have reached a natural conclusion, I will close it now.  Thanks so much @michelodu  and good luck with your work!

closed 2020-08-08 by MartyG-RealSense