realsenseai/librealsense #6979: Saving point cloud as ply including normals

philianeles, 2020-07-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Operating System & Version | Win 10 | 
| Platform                                 | PC |
| SDK Version                            |  2.0 | 
| Language                            | C | 
| Segment			|  Robot | 

I have recently built the librealsense sdk from source.
I am looking for an C++ example for saving point cloud as ply, including normals, would there be one?

MartyG-RealSense, 2020-07-30:
The link below has a C++ script submitted by a RealSense user that they said exports a ply with mesh and faces:

https://github.com/IntelRealSense/librealsense/issues/6411

That RealSense user and another one also took the alternative approach of computing a point cloud with **voxels** instead of normals:

https://github.com/IntelRealSense/librealsense/issues/6802#issuecomment-656807819

If using C++ is not an absolute requirement, the link below has a guide for exporting a ply from the RealSense Viewer and then importing it into a program called **MeshLab** and computing the normals:

https://www.andreasjakl.com/capturing-3d-point-cloud-intel-realsense-converting-mesh-meshlab/

philianeles, 2020-07-31:
Thanks @MartyG-RealSense!
Now I found `save_to_ply `in `librealsense\include\librealsense2\hpp` which should ideally work: 

```
rs2::save_to_ply exporter("Test ", filtered_pc);
exporter.set_option(rs2::save_to_ply::OPTION_IGNORE_COLOR, 0.f);
exporter.set_option(rs2::save_to_ply::OPTION_PLY_MESH, 0.f);
exporter.set_option(rs2::save_to_ply::OPTION_PLY_NORMALS, 0.f);
```

But I don`t see the ply saved in the directory - any idea what could be the problem?

MartyG-RealSense, 2020-07-31:
It looks as though your script is on the right track.  Compared to the C++ script in the link below that is suggested by Dorodnic the RealSense SDK Manager though, your version does not have an **exporter.export_to_ply** statement at the end of it after the exporter.set_option statements.

https://github.com/IntelRealSense/librealsense/issues/6677#issuecomment-648079238

philianeles, 2020-08-03:
after adding that too:
```
rs2::save_to_ply exporter("Test ", filtered_pc);
exporter.set_option(rs2::save_to_ply::OPTION_PLY_BINARY, 0.f);
exporter.set_option(rs2::save_to_ply::OPTION_IGNORE_COLOR, 1.f);
exporter.set_option(rs2::save_to_ply::OPTION_PLY_MESH, 1.f);
exporter.set_option(rs2::save_to_ply::OPTION_PLY_NORMALS, 1.f);
exporter.export_to_ply(filtered_points, colored_filtered);
```
I have this error:
'rs2::save_to_ply::export_to_ply': cannot access private member declared in class 'rs2::save_to_ply'	

Any clues how to avoid this?

MartyG-RealSense, 2020-08-03:
In the **rs2::save_to_ply exporter** line, I'm not sure that "Test " is a valid path for a location and name to save the ply file to.   For example, in the Dorodnic example I quoted, the RealSense user who that case belonged to set up a **path** variable in the following way:

**char path[200];
sprintf_s(path, "F:\\20191024_%d.ply", ind);**

philianeles, 2020-08-03:
also tried that but the error persists (please see below):
[image]

MartyG-RealSense, 2020-08-03:
Comparing your script to the earlier case with Dorodnic's advice that I linked to, it looks as though you should include this in your rs2:: list:

**rs2::pointcloud pc;**

And then set the **pc** definition in save_to_ply:

**rs2::save_to_ply exporter('test.ply', pc);**

I also suspect that by having more than one type of the same rs2:: reference (two rs2::frame and rs2::points) instead of one rs2:: for each type, it may be confusing the program.

philianeles, 2020-08-09:
[image]

Even though I would include `rs2::pointcloud pc` in my `rs2:: list` for saving, the problem is rather I think the `export_to_ply` is a private function in  rs_export.hpp --> the exporter object can't visit it. @dorodnic would you be able to provide some example code how to use the `export_to_ply` including normals?

philianeles, 2020-08-24:
@MartyG-RealSense thank you! the issue that I had before (https://github.com/IntelRealSense/librealsense/issues/6979#issuecomment-671079915) seems to be gone with this new lead however I see this error now: 
[image]

would you have an idea why this error is thrown?

MartyG-RealSense, 2020-08-24:
Assuming that the program got past the **switch** instruction okay, I would guess that it is failing at the **CrtDbgBreak()** instruction.  It looks as though this is a call to check a function called CrtDbgBreak().  Do you actually have a function in the project with that exact name **CrtDbgBreak**() ?

philianeles, 2020-08-24:
I am actually trying in an existing example program (`librealsense2.sln`), inside `rs-post-processing `
here is the whole thing: (it runs without throwing an error but there is no test.ply saved anywhere? from this line: `points_to_save.export_to_ply("test.ply", colored_filtered);  at the very end)`
```
// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2017 Intel Corporation. All Rights Reserved.

#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
#include "example.hpp"          // Include short list of convenience functions for rendering
#include <librealsense2/hpp/rs_export.hpp> // Include RealSense Cross Platform API

#include <map>
#include <string>
#include <thread>
#include <atomic>

#include <imgui.h>
#include "imgui_impl_glfw.h"

/**
Helper class for controlling the filter's GUI element
*/
struct filter_slider_ui
{
    std::string name;
    std::string label;
    std::string description;
    bool is_int;
    float value;
    rs2::option_range range;

    bool render(const float3& location, bool enabled);
    static bool is_all_integers(const rs2::option_range& range);
};

/**
Class to encapsulate a filter alongside its options
*/
class filter_options
{
public:
    filter_options(const std::string name, rs2::filter& filter);
    filter_options(filter_options&& other);
    std::string filter_name;                                   //Friendly name of the filter
    rs2::filter& filter;                                       //The filter in use
    std::map<rs2_option, filter_slider_ui> supported_options;  //maps from an option supported by the filter, to the corresponding slider
    std::atomic_bool is_enabled;                               //A boolean controlled by the user that determines whether to apply the filter or not
};

// Helper functions for rendering the UI
void render_ui(float w, float h, std::vector<filter_options>& filters);
// Helper function for getting data from the queues and updating the view
void update_data(rs2::frame_queue& data, rs2::frame& depth, rs2::points& points, rs2::pointcloud& pc, glfw_state& view, rs2::colorizer& color_map);

int main(int argc, char * argv[]) try
{
    // Create a simple OpenGL window for rendering:
    window app(1280, 720, "RealSense Post Processing Example");
    ImGui_ImplGlfw_Init(app, false);

    // Construct objects to manage view state
    glfw_state original_view_orientation{};
    glfw_state filtered_view_orientation{};

    // Declare pointcloud objects, for calculating pointclouds and texture mappings
    rs2::pointcloud original_pc;
    rs2::pointcloud filtered_pc;

    // Declare RealSense pipeline, encapsulating the actual device and sensors
    rs2::pipeline pipe;
    rs2::config cfg;
[... 140 lines cut ...]
            filtered_view_orientation.yaw += rotation_velocity;
            last_time = curr;
        }
    }

    rs2::pointcloud pc_to_save;
    rs2::points points_to_save;

    points_to_save = pc_to_save.calculate(colored_depth);

    //rs2::save_to_ply exporter("test.ply ", pc_to_save);
    //exporter.set_option(rs2::save_to_ply::OPTION_PLY_BINARY, 0.f);
    //exporter.set_option(rs2::save_to_ply::OPTION_IGNORE_COLOR, 1.f);
    //exporter.set_option(rs2::save_to_ply::OPTION_PLY_MESH, 1.f);
    //exporter.set_option(rs2::save_to_ply::OPTION_PLY_NORMALS, 1.f);
    points_to_save.export_to_ply("test.ply", colored_filtered);

    // Signal the processing thread to stop, and join
    // (Not the safest way to join a thread, please wrap your threads in some RAII manner)
    stopped = true;
    processing_thread.join();

    return EXIT_SUCCESS;
}
```

MartyG-RealSense, 2020-08-25:
At the bottom of the script, most of the point cloud saving code is commented out, preventing it from running.  This includes the definitions for the format that the data will be saved in.

If the ply file was successfully saved, I suspect that it would be saved in the same folder as the script since there is not a path defined for "test.ply".

philianeles, 2020-08-25:
I am not sure how the settings would effect the actual export action (would it not take default settings if custom ones are not provided?) but just in case I have commented them out. Still I do not know why I cannot see the ply in the same folder. As you pointed out, it should save it in the same folder if no path is given - or am I missing something?

MartyG-RealSense, 2020-08-25:
Does it still error if you do this:

**points_to_save.export_to_ply(points_to_save, colored_filtered);**

philianeles, 2020-08-25:
Yes then I throws this error: "no suitable user-defined conversion from "rs2::points" to "const std::string" exists"

MartyG-RealSense, 2020-08-25:
At this point the best way forward would be to get a second opinion on your script from a developer so that we don't go round in circles.   @dorodnic or @ev-mp could you take a look at the ply export script above please to see why a ply file is not being created?   Thanks!

MartyG-RealSense, 2020-09-03:
Reminder:  @dorodnic or @ev-mp could you take a look at the ply export script above please to see why a ply file is not being created? Thanks!

MartyG-RealSense, 2020-09-13:
Hi @philianeles  I apologise that you have not received further comment about your ply export problem.  I have highlighted the case with Intel internally to draw attention to it.

RealSenseSupport, 2020-09-14:
The point cloud is calculated based on the depth frame. The rs-post-processing has an example of how to properly generate the point cloud from the depth frame, colorized the depth and merge it with the point could. Please check:
https://github.com/IntelRealSense/librealsense/blob/master/examples/post-processing/rs-post-processing.cpp#L237

Replacing the colored_depth with the actual depth frame would make your code work. 
The ply file will be generated on exit in the same directory where the program run.

[image]

```
	rs2::frame f;
	if (filtered_data.poll_for_frame(&f))  // Try to take the depth and points from the queue
	{
		colorized_depth = color_map.process(f);     // Colorize the depth frame with a color map
		pc_to_save.map_to(colorized_depth);         // Map the colored depth to the point cloud
		points_to_save = pc_to_save.calculate(f); // Generate pointcloud from the depth data
		points_to_save.export_to_ply("test.ply", colorized_depth);
	}

    // Signal the processing thread to stop, and join
    // (Not the safest way to join a thread, please wrap your threads in some RAII manner)
    stopped = true;
    processing_thread.join();
```

philianeles, 2020-09-21:
Thanks for the tips. There is no error:
[image]

However I still do not see any .ply generated on exit in the same directory where the program runs (or anywhere else):
[image]

philianeles, 2020-09-21:
sorry it works now (somehow not in debugging mode from VS but only running the .exe)! Anyhow, thanks for the help!

closed 2020-09-21 by philianeles

philianeles, 2020-09-21:
@RealSenseSupport and one more question how can I use options like `OPTION_PLY_BINARY`, `OPTION_PLY_NORMALS` or `OPTION_PLY_MESH` with `export_to_ply`?