realsenseai/librealsense #6980: pyrealsense2 on python2 Jetson Xavier NX

GOBish, 2020-07-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | () | 
| Operating System & Version |   { Linux (Ubuntu 14/16/17)   | 
| Kernel Version (Linux Only)    |  ()                                         | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            |  { 2.36 }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

I have been trying to install the pyrealsense2 library on the jetson Xavier nx. I have successfully installed it and used it with python2. (note you still cannot use pip or pip3 to get the library for the jetson).

Please see this issue for background: https://github.com/IntelRealSense/librealsense/issues/6449 
@MartyG-RealSense has been very helpful in trying to figure this out, but neither of us have been able to.

I cannot get the library to install for python3.  Has anyone been able to do it?

Flags I have tried when compiling:
cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true -DFORCE_RSUSB_BACKEND=ON -DPYTHON_EXECUTABLE=... -DBUILD_WITH_CUDA:bool=true 

 cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true -DFORCE_RSUSB_BACKEND=ON -DPYTHON_EXECUTABLE=... -DBUILD_WITH_CUDA:bool=true  -DPYTHON_EXECUTABLE=/usr/bin/python3

 cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true -DFORCE_RSUSB_BACKEND=ON -DPYTHON_EXECUTABLE=... -DBUILD_WITH_CUDA:bool=true  -DPYTHON_EXECUTABLE=/usr/bin/python3.6

No matter what, the libraries always get installed to the /usr/local/lib/python2.7/pyrealsense2 directory.

If I try to import the library using python3, I get the error `undefined symbol: _Py_ZeroStruct`

Has anyone figured out why this may be happening with the Jetson environment? I could really use the help as I have certain other programs that only work with python3 and I need to be able to use realsense along with those programs.

ev-mp, 2020-07-30:
@GOBish , 
If indeed the python binaries are built for v2.7 regardless of `-DPYTHON_EXECUTABLE=/usr/bin/python3.6` and `-DPYTHON_EXECUTABLE=/usr/bin/python3` then CMake probably can't find/use those files and falls back to v2.7.
Did you verify that python3 including its `-dev` package are installed properly? 

If so then you can share a full log for review
```sh
cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true -DFORCE_RSUSB_BACKEND=ON -DPYTHON_EXECUTABLE=... -DBUILD_WITH_CUDA:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3 && make -j$(($(nproc)-1)) && sudo make install
```

GOBish, 2020-07-30:
sorry, that was just a copy paste error in my post for the last line. The actual command I used only had one -DPYTHON_EXECUTABLE.
I will run the cmake as you suggest and post the log shortly.

GOBish, 2020-07-30:
also, 
output from sudo apt-get install python3-dev

python3-dev is already the newest version (3.6.7-1~18.04).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

GOBish, 2020-07-30:
$ cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true -DFORCE_RSUSB_BACKEND=ON -DBUILD_WITH_CUDA:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3 && make -j$(($(nproc)-1)) && sudo make install
-- Internet connection identified
-- Info: REALSENSE_VERSION_STRING=2.36.0
-- Setting Unix configurations
-- Info: Building with CUDA requires CMake v3.8+
-- CUDA_LIBRARIES: /usr/local/cuda-10.2/include /usr/local/cuda-10.2/lib64/libcudart_static.a;-pthread;dl;/usr/lib/aarch64-linux-gnu/librt.so;/usr/local/cuda-10.2/lib64/libcusparse.so;/usr/lib/aarch64-linux-gnu/libcublas.so
-- using RS2_USE_LIBUVC_BACKEND
-- Found PythonInterp: /usr/bin/python3 (found version "3.6.9") 
-- pybind11 v2.2.1
-- Could NOT find apriltag (missing: APRILTAG_INC APRILTAG_LIB) 
-- Unable to find apriltag library, skipping pose-apriltag example
-- Building with TM2
-- Fetching recommended firmwares:
-- D4XX_FW_VERSION: 5.12.6.0
-- SR3XX_FW_VERSION: 3.26.1.0
-- T26X_FW_VERSION: 0.2.0.951
-- L5XX_FW_VERSION: 1.4.1.2
-- http://realsense-hw-public.s3.amazonaws.com/Releases/RS4xx/FW/D4XX_FW_Image-5.12.6.0.bin
-- Download firmware 0;"returning early; file already exists with expected SHA1 hash" for D4XX_FW_Image-5.12.6.0.bin
-- http://realsense-hw-public.s3.amazonaws.com/Releases/SR300/FW/SR3XX_FW_Image-3.26.1.0.bin
-- Download firmware 0;"returning early; file already exists with expected SHA1 hash" for SR3XX_FW_Image-3.26.1.0.bin
-- http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/target/0.2.0.951/target-0.2.0.951.mvcmd
-- Download firmware 0;"returning early; file already exists with expected SHA1 hash" for target-0.2.0.951.mvcmd
-- http://realsense-hw-public.s3.amazonaws.com/Releases/L5xx/FW/L5XX_FW_Image-1.4.1.2.bin
-- Download firmware 0;"returning early; file already exists with expected SHA1 hash" for L5XX_FW_Image-1.4.1.2.bin
-- Configuring done
-- Generating done
-- Build files have been written to: /home/gbenel/librealsense/build
[  1%] Built target fw
[  2%] Linking CXX shared library pybackend2.so
[  8%] Built target realsense-file
[  8%] Linking CXX shared library librealsense2.so
[ 45%] Built target realsense2
Scanning dependencies of target rs-hello-realsense
Scanning dependencies of target rs-software-device
Scanning dependencies of target realsense2-gl
[ 45%] Building CXX object examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o
[ 45%] Building CXX object examples/software-device/CMakeFiles/rs-software-device.dir/rs-software-device.cpp.o
[ 45%] Linking CXX shared library pyrealsense2.so
[ 46%] Building CXX object src/gl/CMakeFiles/realsense2-gl.dir/synthetic-stream-gl.cpp.o
[ 47%] Linking CXX executable rs-hello-realsense
[ 47%] Built target rs-hello-realsense
Scanning dependencies of target rs-capture
[ 47%] Building CXX object examples/capture/CMakeFiles/rs-capture.dir/rs-capture.cpp.o
[ 47%] Linking CXX executable rs-software-device
[ 47%] Built target rs-software-device
Scanning dependencies of target rs-callback
[ 47%] Building CXX object examples/callback/CMakeFiles/rs-callback.dir/rs-callback.cpp.o
[ 47%] Linking CXX executable rs-capture
[ 47%] Built target rs-capture
Scanning dependencies of target rs-save-to-disk
[ 47%] Building CXX object examples/save-to-disk/CMakeFiles/rs-save-to-disk.dir/rs-save-to-disk.cpp.o
[ 48%] Linking CXX executable rs-callback
[ 48%] Built target rs-callback
Scanning dependencies of target rs-multicam
[ 48%] Building CXX object examples/multicam/CMakeFiles/rs-multicam.dir/rs-multicam.cpp.o
[ 48%] Linking CXX executable rs-save-to-disk
[ 48%] Built target rs-save-to-disk
Scanning dependencies of target rs-pointcloud
[ 48%] Building CXX object examples/pointcloud/CMakeFiles/rs-pointcloud.dir/rs-pointcloud.cpp.o
[ 48%] Building CXX object src/gl/CMakeFiles/realsense2-gl.dir/yuy2rgb-gl.cpp.o
In file included from /usr/include/c++/7/backward/strstream:50:0,
                 from /home/gbenel/librealsense/src/gl/yuy2rgb-gl.cpp:19:
/usr/include/c++/7/backward/backward_warning.h:32:2: warning: #warning This file includes at least one deprecated or antiquated header which may be removed without further notice at a future date. Please use a non-deprecated interface with equivalent functionality instead. For a listing of replacement headers and interfaces, consult the file backward_warning.h. To disable this warning use -Wno-deprecated. [-Wcpp]
 #warning \
  ^~~~~~~
[ 49%] Linking CXX executable rs-multicam
[ 49%] Linking CXX executable rs-pointcloud
[ 49%] Built target rs-pointcloud
[ 49%] Built target rs-multicam
[... 425 lines cut ...]
-- Installing: /usr/local/bin/rs-tracking-and-depth
-- Set runtime path of "/usr/local/bin/rs-tracking-and-depth" to ""
-- Installing: /usr/local/bin/rs-convert
-- Set runtime path of "/usr/local/bin/rs-convert" to ""
-- Installing: /usr/local/bin/rs-enumerate-devices
-- Set runtime path of "/usr/local/bin/rs-enumerate-devices" to ""
-- Installing: /usr/local/bin/rs-fw-logger
-- Set runtime path of "/usr/local/bin/rs-fw-logger" to ""
-- Installing: /usr/local/bin/rs-terminal
-- Set runtime path of "/usr/local/bin/rs-terminal" to ""
-- Installing: /usr/local/bin/rs-record
-- Set runtime path of "/usr/local/bin/rs-record" to ""
-- Installing: /usr/local/bin/rs-fw-update
-- Set runtime path of "/usr/local/bin/rs-fw-update" to ""
-- Installing: /usr/local/bin/rs-data-collect
-- Set runtime path of "/usr/local/bin/rs-data-collect" to ""
-- Installing: /usr/local/bin/realsense-viewer
-- Set runtime path of "/usr/local/bin/realsense-viewer" to ""
-- Installing: /usr/local/bin/rs-depth-quality
-- Set runtime path of "/usr/local/bin/rs-depth-quality" to ""
-- Installing: /usr/local/bin/rs-rosbag-inspector
-- Set runtime path of "/usr/local/bin/rs-rosbag-inspector" to ""
-- Installing: /usr/local/bin/rs-benchmark
-- Set runtime path of "/usr/local/bin/rs-benchmark" to ""
-- Up-to-date: /usr/local/lib/libfw.a

Fred3D-tech, 2020-07-30:
a lot of us are using jetson xavier to make mobile devices..
Is it not a time to have a tutorial to make a clean, bullet proof installation ? I really think that is not serious to not take into account the people working on nvidia jetson xavier as it it the best mobile platform to work with realsense products... I have 10 DR400 products and 2 L515 devices.. I really feel very frustrated.

ev-mp, 2020-07-30:
@GOBish , what Cmake version are you using, Is it 3.12+?  If so then the installation path is derived from `${Python_SITEARCH}` variable. [This](https://github.com/IntelRealSense/librealsense/pull/6220/commits/3a4d1e66b8e2088ede3817ba54fa3d0c13b130a7) could be the source of the inconsistency between `-DPYTHON_EXECUTABLE=/usr/bin/python3` and `${Python_SITEARCH}=python2.7`
I'll try to ping @morxa to look into this as well.

GOBish, 2020-07-30:
cmake --version
cmake version 3.10.2

GOBish, 2020-07-31:
Raising the issue of Cmake got me thinking. I updated to a newer version of Cmake and rebuilt everything, and it now works with python3!

Here are the steps I took for updating Cmake on the Xavier:

1.wget http://www.cmake.org/files/v3.13/cmake-3.13.0.tar.gz
2. tar xpvf cmake-3.13.0.tar.gz cmake-3.13.0/
3. cd cmake-3.13.0/
4. ./bootstrap --system-curl
5. make -j6
6. echo 'export PATH=/home/nvidia/cmake-3.13.0/bin/:$PATH' >> ~/.bashrc
7. source ~/.bashrc

RealSenseSupport, 2020-08-06:
@GOBish, thank you for letting us know t that it works after updating cmake to 3.12+. 
We will close this issue unless otherwise. Thank you.

closed 2020-08-06 by RealSenseSupport

Drkstr, 2020-10-21:
Hi @RealSenseSupport @RealSense-Customer-Engineering 

I am trying to use the python wrapper on the NX in python3. These are the steps I used to build the wrapper from source. 
https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 I used librealsense version 2.39 

I have tried all the solutions posted here but I am still getting a " "ModuleNotFoundError: No module named 'pyrealsense2'"" error when I try to import into python3.

Do you guys have any other suggestions for me?

GOBish, 2020-10-21:
Does it work when you try it with just python and not python3?

Drkstr, 2020-10-21:
@GOBish nope doesn't work with python or python3

I set DPYTHON_EXECUTABLE=/usr/bin/python3.6 

I tried importing from inside the build folder. Still no luck. 

```
rakshak@jetsonNx:~/git_repos/librealsense-2.39.0/build$ python
Python 2.7.17 (default, Sep 30 2020, 13:38:04) 
[GCC 7.5.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyrealsense2 as rs
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named pyrealsense2
>>> exit()
rakshak@jetsonNx:~/git_repos/librealsense-2.39.0/build$ python3
Python 3.6.9 (default, Oct  8 2020, 12:12:24) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyrealsense2 as rs
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'pyrealsense2'
```

GOBish, 2020-10-21:
Have you set the pythonpath?

export PYTHONPATH=$PYTHONPATH:/usr/local/lib/python2.7/pyrealsense2

Drkstr, 2020-10-21:
Yup, that was it. I did not have it in my path. 

I set it to /usr/local/lib/python3.6/pyrealsense2 for python3 

Thanks for that :D

IquanKoxinga, 2020-11-12:
@RealSenseSupport @Drkstr @GOBish 
Thank you for sharing your solutions. Currently, I can work with python2 but not with python3. I tried to update the CMake, but it is still not working. What is the next step after the CMake update?

SolidState74, 2020-11-14:
@RealSenseSupport @MartyG-RealSense 

I am having the exact same issue as @IquanKoxinga. I've tried every single tutorial/git I could find to install pyrealsense2 with python3.6. No matter what I try, I cannot get pyrealsense2 installed into my python3.6 path. I have completely run out of options/ideas.

FWIW, I can get pyrealsense2 to build into python2.7, but I need it to work with python3.6 in order to use my D455. Please help!

Drkstr, 2020-11-14:
@IquanKoxinga @SolidState74 what error are you getting? 

Did you complete all the steps I have detailed here without error? https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 

If you have id try trying again with a fresh install of the OS. I know its probably not what you want to hear, but that's what I ended up needs to do during one of my attempts to get the thing working.

MartyG-RealSense, 2020-11-14:
Hi @SolidState74   If you are using a Jetson: as well as the guide of @Drkstr that has worked for a number of people, there is a new user-contributed Jetson guide that was recently submitted and targets Python 3.6 with its PYTHONPATH.

https://github.com/IntelRealSense/librealsense/issues/7722

SolidState74, 2020-11-14:
@Drkstr @MartyG-RealSense 
Thank you for the quick replies. @Drkstr,I had followed your guide yesterday, and everything finished without error, but I couldn't get the pyrealsense2 "stuff" to show up in my python3.6 folder. I am going to try today with a fresh OS flash. I will report back once I have some new info. Thanks again :)

GOBish, 2020-11-14:
@SolidState74 there was a point where the “stuff” didn’t show up in the python3 folder for me as well but  I was still able to use python3 with the correct pythonpath setting

lieuzhenghong, 2020-11-15:
Hi @SolidState74 , let me know if the instructions work for you! I might have missed something out and want to make sure it's correct before making a PR. @MartyG-RealSense

SolidState74, 2020-11-15:
@Drkstr @MartyG-RealSense 

I followed [https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049](url) after a fresh OS flash, and everything is now working! Thanks again for the help on this. A couple of details:

1. I did indeed need to install curl to upgrade Cmake
2. When attempting step 5, I encountered the no CMAKE_CUDA_COMPILER error, fix here: [https://github.com/jetsonhacks/buildLibrealsense2TX/issues/13](url)
3. After fixing above issue, I encountered a missing libxinerama error, which I solved with:
`sudo apt-get install libxinerama-dev`
4. After fixing missing libxinerama issue, I encountered a missing libxcursor error, which I solved with:
`sudo apt-get install libxcursor-dev`
5. After fixing missing libxcursor issue, CMAKE command worked as expected (I omitted the whitespace between the - and DPYTHON_EXECUTABLE)

I was able to complete the rest of the steps without issue, and a test run of a python script in python3 works perfectly.

Thanks again!

SolidState74, 2020-11-15:
> 
> 
> Hi @SolidState74 , let me know if the instructions on #7722 work for you! I might have missed something out and want to make sure it's correct before making a PR. @MartyG-RealSense

@lieuzhenghong I did not end up following your instructions in this most recent attempt, but I did the other day (thank you for putting that together). I didn't run into any issues with your steps. In the end, I had to reflash my OS to get things working.

Drkstr, 2020-11-15:
@SolidState74 glad you got it working :)

MartyG-RealSense, 2020-11-16:
Yes @SolidState74  It's great to hear of your success.   Thanks for the update    :)

IquanKoxinga, 2020-11-18:
@MartyG-RealSense 
I just solved my problem. My problem was the files for python3 were built in a different folder ( not the same as python2.7): 'usr/bin/lib'. Thanks.

MartyG-RealSense, 2020-11-19:
Hi @IquanKoxinga   That's great news - thanks for the update!