realsenseai/librealsense #6981: Choosing points for depth planes/3D lines and calculating the angle between them

nikstp, 2020-07-30:
------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | D435i | L515                                |
| Firmware Version                   | 2.36.0                                                      | 
| Operating System & Version |  Win 10                                                   | 
| Platform                                 | PC                                                           |
| SDK Version                           |  2.0                                                          | 
| Language                               |  C                                                            | 
| Segment			        | others                                                      | 

### Issue Description
Hi,

I'm working with 415/D435i and L515 cameras and my goal is to measure an angle between 2 3D objects (preferably with 1 camera).  It seems that it could possibly be accomplished using the depth_plane command, which returns the normal vector of the plane, or by constructing a cloud-point of the scene and drawing a line through each object--are those reasonable approaches? The problem with those approaches, however, is that before I can draw a line or by getting a depth plane, I need to know the points that I would like to use--is there an easy way to choose points? Would I have to use OpenCV to detect lines and boundaries between objects?

Thank you so much.

dorodnic, 2020-08-05:
Hi @nikstp 
This is rather general question. The SDK does not provide any advanced point-cloud analysis APIs, focusing mainly on 3D data acquisition. PCL and Open3D may be good candidates for shape analysis. That said, SDK tools do have some related functionality that you may find useful. RealSense Viewer has point-to-point measurement functionality and Depth Quality Tool has basic plane fitting. You can look into the implementation and borrow any parts

RealSenseSupport, 2020-08-13:
@nikstp Do you have any other questions with the reply from dorodnic? Looking forward to your update. Thanks!

nikstp, 2020-08-14:
Thank you for your answer, @dorodnic, and thank you for following up with me, @RealSenseSupport!

My question is very general and thus I'm still having some trouble finding what exactly would be of use to me. I'm looking at the implementation of the RealSense Viewer and Depth Quality Tool for the above-mentioned parts. I guess my next question is whether there is a way to limit the areas of where to acquire depth data from? Same applies to processing--if I want to choose a specific area(s) of the frame(s), do I have to specify the location manually (and if so, how would I do it?) or I can do it via "feature extraction"? 

Thank you so much!

saratrajput, 2020-09-09:
@nikstp Could you please tell me where to find the depth_plane command? I'm exploring a similar problem related to finding the surface normal for a given point, but I'm not able to find any inbuilt function.

nikstp, 2020-09-09:
@saratrajput Hi Saraj, you can find the plane fitting* (it isn't called a depth plane per se) function within the source code for the Intel Real Sense Depth Quality Tool:

https://github.com/IntelRealSense/librealsense/blob/d39b5139ce62efffddee4cd11a87e8927ecdfa0c/tools/depth-quality/depth-metrics.h#L62-L63

RealSenseSupport, 2020-12-24:
Sorry for late.
There are many algorithms can help you, I think. But it's not a librealsense related. :-)
for example,
https://stackoverflow.com/questions/28731442/detecting-set-of-planes-from-point-cloud
https://bdataanalytics.biomedcentral.com/articles/10.1186/s41044-018-0035-y

nikstp, 2021-01-26:
Hi, 

I've been looking at this problem again recently. I can see code that I can reuse from some samples provided, but I haven't been really able to understand where the points that are obtained from a frame that are used to construct a point-cloud or a depth frame lie. Is it the rs_vector type array? Ideally, what I'm looking at doing now is being able to choose a few points with my mouse on the depth_frame or a point-cloud and then process the points selected without having to rely on extensive libraries like PCL, since what I want to actually do for processing should be quite simple at the moment. 

Thank you so much!

RealSenseSupport, 2021-12-14:
@nikstp Any other questions about this ticket? Looking forward to your reply. Thanks!

MartyG-RealSense, 2026-10-23:
Case closed due to no further comments received after a significant time period.

closed 2026-10-23 by MartyG-RealSense