realsenseai/librealsense #6982: jetson nano syslog spam tegra-xusb 70090000.xusb: WARN Event TRB for slot X ep Y with no TDs queued?

ethanrublee, 2020-07-31:
```
Device info: 
    Name                          : 	Intel RealSense T265
    Serial Number                 : 	15322110300
    Firmware Version              : 	0.2.0.926
    Physical Port                 : 	2-1.3-8
    Product Id                    : 	0B37
    Usb Type Descriptor           : 	3.1
    Product Line                  : 	T200

Stream Profiles supported by Tracking Module
 Supported modes:
    stream       resolution      fps       format   
    Fisheye 1	  848x800	@ 30Hz	   Y8
    Fisheye 2	  848x800	@ 30Hz	   Y8
    Gyro	 N/A		@ 200Hz	   MOTION_XYZ32F
    Accel	 N/A		@ 62Hz	   MOTION_XYZ32F
    Pose	 N/A		@ 200Hz	   6DOF

rs-enumerate-devices  version: 2.34.0
```
```
$ uname -a
Linux 4.9.140-tegra #1 SMP PREEMPT Thu Jun 25 21:25:44 PDT 2020 aarch64 aarch64 aarch64 GNU/Linux
```
platform: tegra nano development board

Building from source:
```
librealsense$ git show
commit bc64e52c9e128aa82cc9f25a14283bbe2bcf8828 (HEAD)
Date:   Mon Apr 13 13:24:57 2020 +0300
```
Cmake line:
```
cmake -DBUILD_PYTHON_BINDINGS=true -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true ../librealsense/
```
Tried with several variations of force UVC backend, but didn't make a difference.

### Issue Description

dmesg/syslog is spammed with several messages per second while capturing:
```
Jul 31 00:27:21 tractor01 kernel: [ 3379.595766] tegra-xusb 70090000.xusb: WARN Event TRB for slot 6 ep 4 with no TDs queued?
Jul 31 00:27:21 tractor01 kernel: [ 3379.705110] tegra-xusb 70090000.xusb: WARN Event TRB for slot 6 ep 4 with no TDs queued?
Jul 31 00:27:21 tractor01 kernel: [ 3379.813728] tegra-xusb 70090000.xusb: WARN Event TRB for slot 6 ep 4 with no TDs queued?
Jul 31 00:27:21 tractor01 kernel: [ 3379.914566] tegra-xusb 70090000.xusb: WARN Event TRB for slot 6 ep 4 with no TDs queued?
Jul 31 00:27:21 tractor01 kernel: [ 3380.027653] tegra-xusb 70090000.xusb: WARN Event TRB for slot 6 ep 4 with no TDs queued?
```

ev-mp, 2020-08-03:
@ethanrublee , this is a recognized false alarm with kernels prior to 4.15 - http://billauer.co.il/blog/2019/11/xhci-controller-bug-warning. By its nature it is cosmetic and harmless, thus can be safely disregarded.
  
One can follow the advise given in the link and retrofit the fix patch into the kernel.

ethanrublee, 2020-08-03:
Thanks @ev-mp for the info.

RealSenseSupport, 2020-08-03:
@ethanrublee Do you need further help? Can we close this?

ethanrublee, 2020-08-03:
Feel free to close.  I'll post back if actually patch the kernel to suppress the spam.

closed 2020-08-03 by ethanrublee