realsenseai/librealsense #6983: Need some answers regarding placing bulk order of depth camera

kafan1986, 2020-07-31:
Hello Team,

I need to place a bulk order of depth camera and would like to discuss it further with relevant team. If you could kindly direct me to the person or team, so that I can get few of my queries sorted out.

MartyG-RealSense, 2020-07-31:
Hi @kafan1986   You can set a large order quantity on listings on the RealSense Store if you keep clicking the '+' button beside Quantity repeatedly (e.g up to 100, though you can go beyond that).

If you are looking to place a very large order (somewhere between 100+ and a million camera units) then a member of the RealSense sales team may be an appropriate contact for you.

kafan1986, 2020-07-31:
@MartyG-RealSense There is a bulk purchase option at your partner store like "Mouser" having lot size of 30 pcs. I also don't need the stand or the cable provided by intel realsense. So the bulk quantity set looks better from my end. Also, maybe I can get better pricing for the product.

MartyG-RealSense, 2020-07-31:
Okay, if you leave your email address and a real-life name then I can arrange for the details to be sent to a sales representative.  If you would rather keep the details private, you are welcome to use the edit button to remove them once I confirm I have the details.

kafan1986, 2020-07-31:
Debasish Mitra
debasish@synapticsense.com

MartyG-RealSense, 2020-07-31:
Okay I got it, I am arranging for the details to be passed on now.   You can remove the contacs now if you wish.  Thanks!

dorodnic, 2020-08-05:
Hi @kafan1986 
Thank you for your interest in RealSense
I don't think GitHub is the right platform to have this discussion
Store support is probably the more appropriate channel - https://www.intel.com/content/www/us/en/forms/support/intel-realsense-store-inquiry.html

closed 2020-08-05 by dorodnic