realsenseai/librealsense #6984: [Jeston nao] The usb wifi is disconnect when using the realsense D435

WSPeng, 2020-07-31:
D435, pyrealsense2, ubuntu18.04, jetson nano, python3

When I am using the camera to collect pictures, the USB wifi is temporally unavailable (can't ping google.com). May I ask how to solve it?

MartyG-RealSense, 2020-07-31:
Hi @WSPeng   Wi-fi transmissions are based on radio waves, so it is likely not related to camera interference (RealSense 400 Series cameras can interfere with **infrared** devices for example, but are not interfered with in return).  

Given that you are using a Nano, it is more likely that when the camera begins streaming and increases the amount of power it is drawing through the USB port, there is not enough power left for the USB wi-fi adapter to stay operational.  This phenomenon is sometimes seen with Jetson devices such as the Nano, where one camera works fine but attaching a second camera results in a **failed to set power state** error.

devshank3, 2020-07-31:
@WSPeng @MartyG-RealSense  

I too faced the same thing, and the issue was due to power withdrawal for D435i. both USB WiFi and wireless keyboard mouse stopped working when D435i is on.

Then I moved on to the M.2 WiFi card and its solved.

MartyG-RealSense, 2020-07-31:
Thanks for your advice @devshank3 !

WSPeng, 2020-08-01:
> Hi @WSPeng Wi-fi transmissions are based on radio waves, so it is likely not related to camera interference (RealSense 400 Series cameras can interfere with **infrared** devices for example, but are not interfered with in return).
> 
> Given that you are using a Nano, it is more likely that when the camera begins streaming and increases the amount of power it is drawing through the USB port, there is not enough power left for the USB wi-fi adapter to stay operational.
> 
> Are you using a mains electricity powered USB 3 hub with your Nano please, or are the camera and wi-fi adapter plugged directly into the Nano's USB ports?

Thank you for your replay. The camera and wifi adapter are plugged directly into the Nano's USB port. I can have try with the powered USB hub when I have the hub available.

closed 2020-08-01 by WSPeng