realsenseai/librealsense #6986: Module 'pyrealsense2' has no attribute 'pipeline'

Kosta404, 2020-07-31:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  SR300  | 
| Firmware Version                   | 2.36 | 
| Operating System & Version |   Raspberry Pi OS (32-bit) release 2020-05-27  | 
| Kernel Version (Linux Only)    |  5.4.51                                          | 
| Platform                                 | Raspberry Pi/.  |
| SDK Version                            |   legacy / 2.36.0                           | 
| Language                            |  python/unity                          | 
| Segment			|  others                           | 

### Issue Description
I`ve installed pyrealsense through this guide https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435. Before reinstalling the system it worked well. Now it says **"Attribute error module 'pyrealsense2' has no attribute 'pipeline' "**

MartyG-RealSense, 2020-07-31:
Hi @Kosta404  I see that **Raspberry Pi OS** is the new name for the Raspbian OS, and that Raspberry Pi OS should work with all Pi models,

Can you tell me which Pi model you are using please (3 / 3b, 4, etc).  Thanks!

Kosta404, 2020-08-03:
@MartyG-RealSense Hello! Yes, I`m using Raspberry Pi 4.

MartyG-RealSense, 2020-08-03:
Thank you very much.  I recall that the Acrobotic installation guide that you used is quite similar to Intel's official installation guide for the Pi 3 / 3b.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md

I notice that you said that you "reinstalled" the system.  Have you had pyrealsense2 installed on your computer before, please?

Kosta404, 2020-08-03:
@MartyG-RealSense Yes, I`ve had it installed on Pi. But I reinstalled my system on a new SD-card.

MartyG-RealSense, 2020-08-03:
Unfortunately, getting pyrealsense2 to work on a Raspberry Pi can sometimes be an awkward process.   I have a memory of a case where a Pi user installed it successfully, backed up their system to SD card and reinstalled from the card but pyrealsense2 no longer worked.  It may be that your pyrealsense2 setup will have to be recreated the long way around (build librealsense, then build pyrealsense2).

Kosta404, 2020-08-03:
@MartyG-RealSense Can you please tell me how to delete all those installed files? Or just delete librealsense directory? Thank you

Kosta404, 2020-08-03:
@MartyG-RealSense Don`t close the topic please. I`ll write here after reinstall. Thank you

MartyG-RealSense, 2020-08-03:
Okay, good luck!

Kosta404, 2020-08-03:
@MartyG-RealSense Hi again . I`ve reinstalled pyrealsense2 and librealsense packages. And built them again according to the official guide suggested above. 
That didn`t work. I have the same problem, not only pipeline function but all the pyrealsense2  functions are not available.
But if i import pyrealsense2 there is no errors in console.

MartyG-RealSense, 2020-08-03:
Thank you very much for your patience.

In the Pi instructions for building pyrealsense2, it provides two separate instructions for building pyrealsense2 for Python 2 and for Python 3.  Have you tried only using the install instruction that applies to the Python version you have installed on your Pi?  I.e if you have Python 2.7, only use the **Python 2** instruction, and if you have a Python 3 version, only use the **Python 3** install instruction.  But not both of them.

Kosta404, 2020-08-03:
@MartyG-RealSense I`ve installed only for python3.

MartyG-RealSense, 2020-08-03:
I believe that if you only have Python 3 installed and not Python 2 then you can leave the **-DPYTHON_EXECUTABLE** command out of the CMake statement.  Some users have had failed pyrealsense2 builds when including this command.

Kosta404, 2020-08-03:
> I believe that if you only have Python 3 installed and not Python 2 then you can leave the **-DPYTHON_EXECUTABLE** command out of the CMake statement. Some users have had failed pyrealsense2 builds when including this command.

No, I have both python2 and python3 installed by default on my system.

MartyG-RealSense, 2020-08-03:
Okay, so you should include **-DPYTHON_EXECUTABLE** then, as it tells CMake which Python version to install the bindings in.  

For some people, using the version of DYPTHON_EXECUTABLE below has worked (i.e three full stops after the equals sign):

**DPYTHON_EXECUTABLE=...**

Kosta404, 2020-08-03:
@MartyG-RealSense I copied and executed the command including specifying version with that part of command, during installation.

Kosta404, 2020-08-03:
@MartyG-RealSense Maybe the problem is in SDK version. The version I had working before is 2.35, but now it is 2.36. Can you please give me an advise how can I install 2.35?

MartyG-RealSense, 2020-08-03:
If you are building librealsense from a source code folder as the Pi installation instruction page suggests, you can get the source code folder for any SDK version from the "Assets" list at the bottom of each version's information on the SDK **Releases** page:

**2.35.0**

https://github.com/IntelRealSense/librealsense/releases/tag/v2.35.0

**2.35.2**

https://github.com/IntelRealSense/librealsense/releases/tag/v2.35.2

Download the source code zip file and extract the **lbrealsense** folder from it, and then do the usual process of mkdir build and cd build, and using the CMake statement.

[image]

Kosta404, 2020-08-03:
> If you are building librealsense from a source code folder as the Pi installation instruction page suggests, you can get the source code folder for any SDK version from the "Assets" list at the bottom of each version's information on the SDK **Releases** page:
> 
> **2.35.0**
> 
> https://github.com/IntelRealSense/librealsense/releases/tag/v2.35.0
> 
> **2.35.2**
> 
> https://github.com/IntelRealSense/librealsense/releases/tag/v2.35.2
> 
> Download the source code zip file and extract the **librealsense** folder from it, and then do the usual process of mkdir build and cd build, and using the CMake statement.
> 
> [image]

Ok, Thank you for assistance. I`ll try and reply tomorrow.

MartyG-RealSense, 2020-08-03:
Thanks again!

MartyG-RealSense, 2020-08-03:
Hi again @Kosta404   A pyrealsense2 user on the Intel Support version of the RealSense forum who was also using the **acrobotic** guide had the **Module 'pyrealsense2' has no attribute 'pipeline'** error too and posted the solution that worked for them.

https://support.intelrealsense.com/hc/en-us/community/posts/360048396854/comments/360012486393

Kosta404, 2020-08-04:
@MartyG-RealSense Hello! I installed librealsense v2.35.2 and all started working. I guess there is some problem with the latest version of librealsense v2.36.0. Thank you for the provided assistance!

closed 2020-08-04 by Kosta404

MartyG-RealSense, 2020-08-04:
Thanks very much for letting us know about your success - great to hear!

pilipili-vr, 2020-08-27:
I have the same issue and tried the solution of installing v2.35.2, but without any success. I still get the error `module 'pyrealsense2' has no attribute 'pipeline'`
Realsense-viewer is working correctly but now I want to use it in Python. Platform is Rpi 4 and I also followed the tutorial as OP.

MartyG-RealSense, 2020-08-27:
Hi @pilipili-vr   Installing the pyrealsense2 wrapper on Pi tends to be unpredictably problematic.  It either goes fine first time or you encounter problems.  It sounds as though you are using the **acrobotic** guide in the link below.  Is that correct,please?

https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435

My recollection of the **module 'pyrealsense2 has no attribute 'pipeline'** error is that it is related to script programming rather than an installation error (where you might instead encounter an error such as **no module named pyrealsense2**).

Does that resemble your situation, please - that you only get the error when running a pyrealsense2 script?

If so, did you set up the pipeline in your pyrealsense2 script in that way that the example program on the front page of the wrapper documentation suggests:

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#streaming-using-rspipeline

import pyrealsense2:

**import pyrealsense2 as rs**

And then define 'pipeline':

**pipeline = rs.pipeline()**

Because pipeline is made equal to RS.pipeline, it depends on 'rs' being set up in the first line by '*import pyrealsense2 as rs'

pilipili-vr, 2020-08-27:
It works when running in terminal. I was using Thonny and that's where I got the error, but running it in terminal everything seems just fine. That's why I removed my comment, but thanks anyway. I did indeed follow the Acrobotic guide.

MartyG-RealSense, 2020-08-27:
Great news that it is working in the terminal for you - thanks very much for the update!