realsenseai/librealsense #6987: How to understand the date which get from IMU calibration?

1556260456, 2020-07-31:
I got some dates after IMU calibration  after I follow the calibration step from  RealSense_Depth_D435i_IMU_Calibration,But I can not understand them because they were not written in the document. So,is there any way to understand them?
the date is like this:

[-0.00640009 -0.00022228 -0.00353566]
[1000 1000 1000 1000 1000 1000]
using 6000 measurements.
[[ 1.02251227  0.01179027 -0.03490604]
 [-0.00237312  1.02114746  0.05023271]
 [-0.00634576 -0.01135111  1.01233392]
 [ 0.04098671  0.49907973  0.18480323]]
residuals: [   8.80501604    2.88540103  203.73583407]
rank: 4
singular: [ 444.33796842  428.44206506  420.73433422   77.1061025 ]
norm (raw data  ): 9.636864
norm (fixed data): 9.804282 A good calibration will be near 9.806650

MartyG-RealSense, 2020-07-31:
I considered carefully how best to explain the numbers, as there are not many detailed references about their meanings.  Most of the values except for the last couple of lines in the list are representative of the sensitivities and biases in the accelerometer and gyroscope data.  For the most part though, you do not need to be too concerned with them.

The last couple of lines in the listing are the most important.

**norm (raw data ): 9.636864
norm (fixed data): 9.804282 A good calibration will be near 9.806650**

The scientific value of the Earth's gravity force is 9.81 (newtons).  So we ideally want the IMU's calculation of acceleration under gravity to be as close to **9.8** as possible in order to be confident in the results that it is providing.  

A problem though is that the IMU component does not have an internal calibration mechanism, and so the **raw** accel value may be something like 9.4 to 9.6.  Another effect of an uncalibrated IMU is that its acceleration values may not be near zero even when the camera is at rest.

To adjust the accel values to take account of known inaccuracies in the IMU, a **motion correction** algorithm can be applied to the accel values, resulting in a corrected - **fixed** - value that is closer to the ideal target of 9.8.      

As the norm in your results above is **9.804282**, it looks like a healthy calibration.

1556260456, 2020-08-01:
HI,very happy to received your reply!
I can understand the  norm (fixed data): 9.804282, what I try to know is the intrinsic parameters of IMU,because  I really need them when I learning VIO. Form the  RealSense_Depth_D435i_IMU_Calibration.pdf, I know we can get Scale factor,Bias ,Off-axis（accelerometer）and another Bias (ayroscope)  using rs-imu-calibration.py . so ,what makes me really confused is that how to get the intrinsic parameters from calibration result. especially I am trying to guess the meaning of  the fir st 3x1 vector and a 4x3 matrix. 
[image]
Best wishes!

MartyG-RealSense, 2020-08-01:
It is difficult to obtain ideal values for the matrices, because the aim is for the camera to be perfectly horizontal or perfectly vertical during each of the positioning steps or calibration but it is hard to achieve that in the real world.  So the results may be negatively affected if the camera orientation in a particular positioning stage during calibration is not perfectly level.

An example of the possible biases in the matrices that may be generated if the camera is not perfectly level is given in the link below.

https://github.com/IntelRealSense/librealsense/issues/5177#issuecomment-550285055

The link below may provide useful advice on better understanding the factors involved in calibration:

https://github.com/IntelRealSense/librealsense/issues/5363#issuecomment-563421712

My understanding is that once an IMU calibration has been written to the camera, you can use **rs-enumerate-devices -c** to view those intrinsics.

closed 2020-08-15 by MartyG-RealSense