realsenseai/librealsense #6988: L515 Python Enable Stream Configs

lesreaper, 2020-07-31:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D4355/L515  | 
| Firmware Version                   | (1.04) | 
| Operating System & Version |   {Jetson Nano Ubutnu 18.04  |                                         
| SDK Version                            |  2.36                          | 
| Language                            |  {Python/PyRealSense }                         

### Issue Description
<Describe your issue / question / feature request / etc..>

I'm attempting to connect my L515 using similar code to what I used for the D435.

```
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)
config.enable_stream(rs.stream.infrared, 1, 640, 480, rs.format.y8, 30)

# Start streaming
profile = pipeline.start(config)
```

Getting a `RuntimeError: Couldn't resolve requests`, which I'm sure means my configs aren't correct. I adjusted the framerates, but everything else seems to match from the `realsense-viewer` settings (which runs perfectly).

What am I missing here?

WOHTECH, 2020-07-31:
Try removing the "1"  in your line "config.enable_stream(rs.stream.infrared, 1, 640, 480, rs.format.y8, 30)"

aurotripathy, 2020-08-02:
Brand new L515. 
None of the python examples with "profile = pipeline.start(config)" work.

```
Traceback (most recent call last):
  File "align-depth2color.py", line 25, in <module>
    profile = pipeline.start(config)
RuntimeError: Couldn't resolve requests

```
Similar compiled functionality work fine.

aurotripathy, 2020-08-02:
The below worked for the LiDAR cam L5151
So it appears that the examples have not been updated for the higher resolutions?

```
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 1024, 768, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)

# Align objects
align_to = rs.stream.color
align = rs.align(align_to)

# Start streaming
profile = pipeline.start(config)
```

RealSenseSupport, 2020-08-20:
Hi,

We are able to run the align python example below (with the higher resolution modification) successfully. Are you still unable to on latest?

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py

BIueMan, 2020-08-27:
it' seems like 640, 480 size for rs.stream.color is not supported with.
what size do supported for the L515?

RealSenseSupport, 2021-01-03:
Hi,

Will you be needing further help with this? You can run rs-enumerate-devices to get a full listing of supported resolutions.

*If we don’t hear from you in 7 days, this issue will be closed.*

Thanks

closed 2021-01-11 by RealSenseSupport

sunilsai7404, 2021-01-18:
I have the same issue and still not able to solve it can any one help with it.