realsenseai/librealsense #6989: Get v-disparity map

Ted-EE, 2020-07-31:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   |  05.12.03.00 | 
| Operating System & Version |  Win10 64-bit  | 
| Kernel Version (Linux Only)    | N/A | 
| Platform                                 | PC |
| SDK Version                            | 2.0                      | 
| Language                            | python                        | 
| Segment			| ground plane detection and removal using v-disparity map                      | 

### Issue Description
Hi, I would like to get v-disparity map using D415. 
At first, it seems that I need to obtain the disparity map, is there any syntax for me to get the disparity map? 
Then I am not sure on the algorithm from disparity map to v-disparity map
I would appreciate it if you provide me some suggestions or examples on the generation of disparity map and v-disparity map.

Thanks!

MartyG-RealSense, 2020-07-31:
Hi @Ted-EE   There is not a lot of information about v-disparity for librealsense, unfortunately.  Another RealSense Python user already tried it.

https://github.com/IntelRealSense/librealsense/issues/4189

As a starting-point reference for creating a disparity map in Python, there is a tutorial on the OpenCV website for creating one with Python and OpenCV.

https://docs.opencv.org/3.4/dd/d53/tutorial_py_depthmap.html

An alternative in Python for ground plane detection and segmentation may be the PCL library;s **SACSegmentation** class.

https://github.com/IntelRealSense/librealsense/issues/6338#issuecomment-623278392

In the comment below the one that I linked to above, a RealSense team member also makes reference to a plane-fit algorithm being added to the **rs-depth-quality** tool.

https://github.com/IntelRealSense/librealsense/issues/6338#issuecomment-623350905

Ted-EE, 2020-08-01:
Hi @MartyG-RealSense Thanks for assistance.

I would like to find an approach to convert depth map into disparity map. In this case [https://docs.opencv.org/3.4/dd/d53/tutorial_py_depthmap.html](url), disparity = focal * baseline / depth. But in my own programme, some value of depth map are zero, therefore gives the error RuntimeWarning: divide by zero encountered in true_divide.

Ted-EE, 2020-08-01:
Just adding a supplement, I would like to perform ground removal on the depth image, that means the ground data will be removed on the depth map.

Thanks

MartyG-RealSense, 2020-08-01:
The subject is somewhat outside of my programming knowledge unfortunately.  The link below may provide some useful insights about another way to generate a custom disparity map with OpenCV though.

https://github.com/IntelRealSense/librealsense/issues/2938#issuecomment-448966295

Numerous other approaches to creating a disparity map with Python can be found by googling for the precise search term below, including the quotation marks:

**opencv "disparity map" "python"**

A few examples of results from such a search include:

https://stackoverflow.com/questions/58113312/calculate-depth-disparity-map-using-opencv

https://github.com/amitthere/stereo-vision-and-segmentation

https://www.geeksforgeeks.org/python-opencv-depth-map-from-stereo-images/

http://amroamroamro.github.io/mexopencv/opencv/stereo_match_demo.html

Ted-EE, 2020-08-01:
Thanks for your help. A quick question, how long does it take to implement the segmentation of the ground plane using PCL library's SACSegmentation class each frame?

MartyG-RealSense, 2020-08-01:
Like most things, how long it takes probably depends on how much previous experience you have with a tool.  If you are proficient in PCL then you might be able to assemble a SACSegmentation test program within a matter of hours.  Otherwise, much longer.  A Python / OpenCV based solution will therefore likely be faster and less headache-inducing if those are areas that you are skilled in.

Ted-EE, 2020-08-01:
I think I may express in a wrong way. What I meant was that what is normal latency of segmentation of the ground plane each frame if applying SACSegmentation.

MartyG-RealSense, 2020-08-01:
I couldn't find any information about latency in regard to SACSegmentation.  It did prompt my memory about using a Python point cloud processing library called **pyntcloud** to calculate ground planes though, as it supports ground plane detection with RANSAC.

https://github.com/daavoo/pyntcloud

https://github.com/kujason/avod/issues/90#issuecomment-412433147

Ted-EE, 2020-08-02:
Many thanks for your suggestion, I will give a try!

MartyG-RealSense, 2020-08-15:
Do you still require assistance with this case, please?   Thanks!

Ted-EE, 2020-08-16:
Nope, thanks!

MartyG-RealSense, 2020-08-16:
Thanks for the update!

closed 2020-08-16 by MartyG-RealSense