realsenseai/librealsense #6990: Should I be able to use rs2::pointcloud with decimated depth?

mbechard, 2020-07-31:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.04.01.02 | 
| Operating System & Version |   {Win10  | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | |
| SDK Version                            |  2.36.0                           | 
| Language                            |  C++ }                          | 
| Segment			|  AR                  | 

### Issue Description

Even in the sample 'pointcloud' app, if I create a decimation filter and set it up with a magnitude of 2, I get a crash when I pass the decimated depth into process() for the pointcloud. The crash occurs deep in the librealsense .dll.
Is this expected to work?
It seems like it sometimes works in my app, but not consistently.

Thanks

ev-mp, 2020-08-03:
@mbechard , the post-processing filters are mostly* agnostic to the source as long as the input formats agree. (*Doesn't apply for disparity_domain).  
There should be no issue to use L515-produced and decimated depth frames for pointcloud generation. Here I [slightly changed the code](https://gist.github.com/ev-mp/f9ac728826d7a7acb07ff50367ad25ba) to make the decimation appear with L515 in the Viewer:
[image]

mbechard, 2020-08-04:
I tried using the latest master and this code crashes using the rs-pointcloud example, using an L515.
https://gist.github.com/mbechard/9761f9617dbd543af4f7ea3c05d857c6
Any ideas?

ev-mp, 2020-08-04:
@mbechard hello, I managed to reproduce the error and understand what went wrong - the issue is specific to pointcloud generation with L515 with RGB texture mapping turned on. 
This is also the reason that it does not affect the above scene - the generated pointcloud is not textured.

We'll handle it internally and update as the fix becomes available.
Note that you can still generate decimated 3D pointclouds if you stick to the following limitations:
- Set Depth resolution 1024X768 
- Set Decimation scale (magnitude factor) to be 2,4 or 8.

```cpp
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_DEPTH, -1, 1024, 768, RS2_FORMAT_Z16, 30);
    cfg.enable_stream(RS2_STREAM_COLOR);
    // Start streaming
    pipe.start(cfg);
```

RealSenseSupport, 2020-09-01:
Hi @mbechard 

Is it safe to close this? Were you able to try with the above fix?

Thanks

*fyi if we don’t hear from you in 7 days, this issue will be closed.*

mbechard, 2020-09-03:
Yep, looks like it. Thanks!

PR #7131 (2020-08-20): Occlusion bug fix
Tracked on: RS5-8522
Fixes #6990
closed 2020-09-03 by mbechard