realsenseai/librealsense #6991: T265 not detected in Windows10 but works in RSviewer

ricardo-jimenez, 2020-07-31:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 } | 
| Firmware Version                   | (0.2.0.926) | 
| Operating System & Version |   {Win 10}  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.<33>.<1> }                          | 
| Language                            |  {python}                          | 
| Segment			|  {others }                          | 

### Issue Description
T265 device is detected by RealSense Viewer and streams data within the viewer but it does not work when operated from a python script. We get the following message: 'Could not open device Unable to create USB device' The device is not identified in the Windows Device manager either.  We have tried the workarounds described in other posts ( power up usb hub first, connect directly to PC, etc) but none has fixed the issue. D415 and D435 devices work fine in the same PC. 

Any help to fix this issue? thanks

VladMVLX, 2020-08-02:
Having the same problem with T265 on Win10 using 2.36 version of sdk, pre release version 2.37 not detecting T265 even in RealSense Viewer

ricardo-jimenez, 2020-08-02:
I upgraded to v2.36, **sometimes** the camera is recognized in the Realsense Viewer **sometimes** it is not recognized.  
Also, the camera appears in the device manager as a USB device, **sometimes** as Intel RealSense (TM) Tracking Camera T265 **sometimes** as Movidius something. It does not work from a python script. This is weird.

VladMVLX, 2020-08-02:
It seams to be working for me when I downgraded to 2.31, but yeah it is super odd.

kenfast, 2020-11-23:
I am having the same issue on a number of Windows 10 machines.  It was working for a while on a Windows 10 Dell tablet, but after some OS updates last week it seems to have stopped working.  Version 2.38.1 of the RealSense viewer does not recognize the device.  It shows up in Device Manager as a USB device, Movidius, but never shows up as a camera.  I also tried v2.31.0 of the viewer, but that didn't work either.

GiovanniCreal, 2022-09-06:
same here, we are desperate... any evolution?

AAslihak, 2022-10-08:
I am also experiencing the same problem. I can get frames within Intel Realsense SDK v2.51; however, T265 camera is not detected from Matlab nor python scripts. I cannot display it in device manager as well. Could you please help us @dorodnic  ?

closed 2026-10-23 by MartyG-RealSense