realsenseai/librealsense #6992: d455 rgb camera not working

BryanChrisBrown, 2020-08-01:
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | (05.12.06.00) | 
| Operating System & Version |   {Win 10  | 
| Platform                                 | PC  |
| SDK Version                            | 2. 36.0                       | 

### Issue Description
Upon plugging the realsense d455 in and opening intel realsense viewer, it does not capture the rgb camera. it simply states "frame not detected" and errors out. the depth feed does work though. 

Does this mean my camera is defective?

MartyG-RealSense, 2020-08-01:
Hi @BryanChrisBrown   RealSense camera hardware faults are extrmely rare.  The problem is more likely to be related to the USB system.  

Are you using the official short USB cable supplied with the camera please, or your own choice of cable that is longer?  Also, are the numbers '2.1' present beside the camera name at the top of the RealSense Viewer's options side-panel?  If you have the camera plugged into a USB port, this would indicate that the camera is being mis-identified as being on a USB 2.0 connection, which provides the camera with more limited capabilities than when it is on a USB 3 connection (typically denoted by the vaue '3.2' beside the camera name).

I note that you also receive connection errors when using RGB in the Windows camera app, which also lends weight to it being a USB related issue and not a software one with the RealSense SDK (which would not be used by the Windows camera app)..

BryanChrisBrown, 2020-08-01:
Hi Marty, 

I am using the included cable, and it is showing up as usb 3.2, here's a screenshot of the realsense viewer. 
[image]

BryanChrisBrown, 2020-08-01:
[image]
These are the errors I get when first booting up the realsense viewer with the camera detected

BryanChrisBrown, 2020-08-01:
I've also tried using my oculus link optical usb cable, and had the same issue.

MartyG-RealSense, 2020-08-01:
Thank you very much for the images.  May I also please confirm that you installed the full RealSense SDK from the **Releases** page (using the 791 mb **Intel.RealSense.SDK-WIN10-2.36.0.2034.exe
** installer file) and then launched the RealSense Viewer using the launcher icon for the Viewer that the installation places on the Windows desktop?

https://github.com/IntelRealSense/librealsense/releases/

[image]

BryanChrisBrown, 2020-08-01:
I am using that version. However I am going to reinstall it just to be sure.

 I’m also not sure what this “button” is on the back of the camera 
[image]

MartyG-RealSense, 2020-08-01:
It is a cover for the hardware sync pins that are used to connect multiple cameras together so that their timing can be synchronized.

BryanChrisBrown, 2020-08-01:
[image]
I get the following set of errors after reinstalling the 2.36.2034 sdk.

MartyG-RealSense, 2020-08-01:
SDK 2.36.0 is the appropriate one for the D455, as it is the version where support for the D455 was introduced.  You are also using the most recent firmware.

I received my own D455 this week, so have just unboxed it for the first time to test it with SDK 2.36.0's Viewer.  I tested the RGB with both USB 3 and USB 2 ports and experienced no errors.

I see that in your supplied image you have Stereo, RGB and Motion Module all enabled.   Does RGB still not work if only RGB only is enabled?

BryanChrisBrown, 2020-08-01:
it appears that rgb and depth work now (this was not the case the last time I tried), but as soon as I turn on the motion module the realsense viewer fails to refresh the rgb stream.

MartyG-RealSense, 2020-08-01:
Enabling the Motion Module after enabling RGB and depth sometimes results in errors.  It is not always clear why this occurs.  It could simply be that a particular computer struggles to have all three stream types active at the same time.

Does the Motion Module generate errors if only it is enabled and RGB and Depth are off (red)?

BryanChrisBrown, 2020-08-01:
it currently appears that turning on the motion module causes the rgb camera to no longer process properly. 

After turning off the rgb camera and stereo camera, I turned on the motion module, and got the expected result. 

however, when I turned the rgb camera back on, it presented me with this screen. 
[image]

MartyG-RealSense, 2020-08-01:
I recalled that I had heard of a similar case recently with the RealSense Viewer where RGB stopped working when the Motion Module was enabled.  Like yourself, that RealSense user also experienced the log error **Exception was thrown during user processing callback!  null pointer passed for argument "frame"**

They reported that most of the time, they most often had problems when enabling stereo first and then RGB secondly.  There was not a clear solution at the conclusion of that case though.

https://support.intelrealsense.com/hc/en-us/community/posts/360048388253/comments/360012195253

BryanChrisBrown, 2020-08-01:
It seems like my problem is solely related to the motion module being enabled. I was able to enable stereo and then rgb successfully

MartyG-RealSense, 2020-08-01:
As the D455 is a very new model whose pre-orders have only just shipped, I cannot rule out the possibility that there is a bug in the 2.36.0 RealSense Viewer program that affects the D455 particularly.  

It may be worth seeing if the camera can work with the **rs-motion** example program that was designed for the D435i's IMU.  You can find a pre-made launchable version of the program by going to **C: > Program Files (x86) > Intel RealSense SDK 2.0** and launching the file **rs-motion**

A fast way to get there is to right-click on the Viewer's desktop launch icon and select **Open file location** from the menu.

BryanChrisBrown, 2020-08-01:
I've recorded a video of what happens with rs-motion. 

it looks like the imu spins rapidly and then slows down and works properly. It will be available at this link once the video is done processing: https://drive.google.com/file/d/1Jvd_KiEa-nqdyt8ZXQfG36Q5fFF7NxP7/view?usp=sharing

MartyG-RealSense, 2020-08-01:
I have not personally tested **rs-motion** with my D455 yet but the behavior that you describe is typical for the D435i, so it sounds as though the IMU is behaving appropriately outside of the Viewer.

BryanChrisBrown, 2020-08-01:
I'll also note that I tried using the viewer with the t265 and the d455 simultaneously, the t265 is plugged into its own dedicated usb 3.1 adapter card. the d455 is plugged in to a dedicated usb c 3.1 port. 

after enabling the t265 in the viewer the rgb stream from the d455 camera errored out. If there are any error logs I can provide to help solve these issues please let me know

MartyG-RealSense, 2020-08-01:
Thanks for the extra detail.   I am not able to handle T265 related support cases though as somebody else is responsible for those, so it is best to keep the conversation focused on 400 Series cameras models.  :)

The link below explains how to set up logging in the Viewer and set a path for the log to be saved at.

https://github.com/IntelRealSense/librealsense/issues/5851

MartyG-RealSense, 2020-08-01:
You have given me an idea though.  If you run the **rs-multicam** example then it will display a series of stream panels for each camera type attached.  I just tested it successfully with D455.  If it can show panels from both D455 and T265 on the same screen then it would lend further weight to your problem being related to the Viewer program specifically.

BryanChrisBrown, 2020-08-01:
[librealsense.log](https://github.com/IntelRealSense/librealsense/files/5010883/librealsense.log)

BryanChrisBrown, 2020-08-01:
rs-multicam is working properly.

MartyG-RealSense, 2020-08-01:
Thanks for the log.   It didn't reveal anything new though, just the previous error generating constantly.

 **01/08 11:56:39,058 ERROR [15024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Error occured during execution of the processing block! See the log for more info
 01/08 11:56:39,092 WARNING [15024] (rs.cpp:1089) null pointer passed for argument "frame"**

BryanChrisBrown, 2020-08-01:
Here's a video of multiple cameras working properly in rs-multicam (2 t265 + 1 d455) definitely seems like it's an issue with the realsense viewer in particular.  https://drive.google.com/file/d/1Hf-qTTa3B1-8dIPBIkL5rWdi9YbkMcyo/view?usp=sharing

Thanks for helping me debug this. I'm glad to know that my camera isn't broken.

MartyG-RealSense, 2020-08-01:
Great video, thanks!

It is a difficult situation, as you are clearly experiencing abnormal behavior in the Viewer but until it can be replicated by other D455 users or by RealSense team members then there is not sufficient grounds to report it as a bug.  

If you feel experimental then you could try the **2.37.0** test version of the SDK (available from the top of the Releases page by expanding open the 'Assets' link).  Or you could wait and see if other new D455 users report similar effects in the coming week as they try out their pre-order cameras.

BryanChrisBrown, 2020-08-01:
same errors with 2.37

MartyG-RealSense, 2020-08-01:
Thanks very much for testing it and providing feedback.

ev-mp, 2020-08-02:
@BryanChrisBrown , it is a [known bug](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#known-issues) in v2.36 for the Viewer compiled with `BUILD_WITH_OPENVINO` flag. You've probably encountered it when running the tool from the Installer as this option is embedded in Win10 distribution package.

To workaround it disable post-processing for the RGB sensor or use the stand-alone Viewer from [v2.36 Release](https://github.com/IntelRealSense/librealsense/releases/download/v2.36.0/Intel.RealSense.Viewer.exe).

[image]

To reconfirm - the issue is isolated to the Viewer tool and does not affect SDK's functionality.

MartyG-RealSense, 2020-08-09:
Case closed due to no further comments received and solution provided.

closed 2020-08-09 by MartyG-RealSense