realsenseai/librealsense #6993: Get the wrong depth if diagonally forward.

Noctstar, 2020-08-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            | python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hello.

Im trying to measure depth of object by realsense d435.
But I've got a problem. The depth of an object placed directly in front of realsense can be measured correctly. However, if it is diagonally in front, it will be a little lower.
E.g) I want it to be 3m, but I got 2.5m.

Is there anyone could teach me how to modify this problem?
Sorry for my bad English.

MartyG-RealSense, 2020-08-01:
Your English is very good,, so don't worry.

It sounds as though that when you say that the object is diagonal, you mean that it is in front of the camera (not to the side of it) but is at a lower position, such as on the floor.  And that measurement accuracy has greater inaccuracy in this position than when the object is directly at the front and center of the camera.  Is this correct please?

If it is, you are certainly not the first person to experience greater measurement error when measuring at an angle.  The advice in the link below about improving accuracy with methods such as depth to color alignment may be a useful starting point for solving your problem.

https://github.com/IntelRealSense/librealsense/issues/2523#issuecomment-430246582

Noctstar, 2020-08-01:
Thanks for the link! @MartyG-RealSense

closed 2020-08-01 by Noctstar

MartyG-RealSense, 2020-08-01:
You are very welcome.   I'm glad I could help!