realsenseai/librealsense #6994: YUYV C04

amy5231, 2020-08-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2020-08-01:
Hello @amy5231   The man distance reading code is the **Hello World** C# example placed inside the brackets of a button-click event.  So clicking the button runs the Hello World program to read the distance.   I cannot see anything obviously wrong in that code section.

I am not certain that you need to open the default camera with **new Capture(0)** though.  The camera will be activated in the button click event code by **pipe.start()**. 

********************

Hello @ amy5231人工遠距離閱讀代碼是Hello World C＃示例，位於按鈕單擊事件的括號內。 因此，單擊按鈕將運行Hello World程序以讀取距離。 在該代碼部分中，我看不到任何明顯錯誤的內容。

我不確定您是否需要使用新的Capture（0）打開默認相機。 將通過pipe.start（）在按鈕單擊事件代碼中激活攝像機。

dorodnic, 2020-08-04:
Hi @amy5231 
My understanding is that starting pipeline with default parameters starts RGB camera, which is also in use from a different API. I second @MartyG-RealSense 's response and believe the issue can be resolved. 
Please comment if there is anything else we can help with?

closed 2020-08-05 by dorodnic