realsenseai/librealsense #6995: Camera not detected on Linux

gywhitel, 2020-08-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 1.1.2.realsense-1.3.14| 
| Operating System & Version |   Linux (Ubuntu 18.04.1 LTS)   | 
| Kernel Version (Linux Only)    |  5.0.0-27-generic                                    | 
| Platform                                 | PC  |

### Issue Description
<Describe your issue / question / feature request / etc..>
The D415 camera works fine on Win10 on my laptop.
But the realsense-viewer cannot detect the camera when running on Ubuntu 18.

```
 modinfo uvcvideo | grep "version:"
version:        1.1.2.realsense-1.3.14
```
```
rs-enumerate-devices 
No device detected. Is it plugged in?
```
Yet using `lsusb`, I do found the device`Bus 002 Device 005: ID 8086:0ad3 Intel Corp. `

MartyG-RealSense, 2020-08-02:
Hi @gywhitel   Which version of the RealSense SDK are you using, please?

Also, **1.1.2.realsense-1.3.14** is not the camera firmware version, it is the version of the Linux **uvcvideo** kernel module.

dorodnic, 2020-08-02:
Another frequent item to check - does it work with `sudo`? If yes, it can be permission issue, see documentation on udev-rules in Linux install guide

gywhitel, 2020-08-02:
Sorry, the version of different packages is confusing....
It's v2.36.0.
I upgraded my kernel, purged the SDK, and reinstalled it.
The viewer can detect the camera now.
Thank you.

MartyG-RealSense, 2020-08-02:
@gywhitel  Great news, thanks for the quick update.  Do you have any further problems that you know of right now, or is okay to close this case?  Thanks!

gywhitel, 2020-08-02:
No problems for now. Thank you! 
It's ok to close this issue.

MartyG-RealSense, 2020-08-02:
Thanks again!

closed 2020-08-02 by MartyG-RealSense