realsenseai/librealsense #6996: switching primary camera in depth estimation algorithm

blakat360, 2020-08-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435 | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |   Ubuntu  | 
| Kernel Version (Linux Only)    |  20.04.1 LTS                                         | 
| Platform                                 | Laptop |
| SDK Version                            |  { legacy / 2.36.0 }                          | 
| Language                            |  C++                          | 
| Segment			|  Other                          | 

When I put my arm up to the camera, the area to the right of it no longer has depth data. I understand that this is due to the fact that one of the cameras is considered the primary camera and the algorithm then tries to find features that that camera sees in the other camera feed.

My application requires that that area retains depth information and I can wait for 2 frames from the camera before processing. Hence, I wish to take one frame, switch the primary camera (thus moving the 'depth shadow' to the left of the object) and repeat.

What code snippet would let me do this?

MartyG-RealSense, 2020-08-02:
Hi @blakat360  If the area to the right of the arm that you are describing is like a black silhouette around it, the phenomenon that you are describing may be **occlusion**.  Because the depth and color cameras are in different physical positions, they cannot see the all the same parts of the scene.

One way to avoid this may be to use the left IR imager as the RGB source instead of the color RGB imager., since IR is aligned perfectly with depth  Though on the D435 model, the IR imagers are both monochrome (the D415's left imager can output RGB color) so this may not be a suitable solution if your project requires a full-color image.

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#section-use-the-left-color-camera

blakat360, 2020-08-02:
I've now read the above but there's no snippet or link to an example to show how to actually implement this.

The section only has a snippet on how to apply colour correction to remove the IR pattern.

MartyG-RealSense, 2020-08-02:
Are you uaing **cfg** (config) statements to set up the streams in your application, please?  If so, then instead of defining a color stream, you would access the color stream of the left IR imager by defining an infrared stream but giving it the **format** for the IR color stream.  For example, to define a 1280x720 color stream (BGR8 format) from the IR imager with a 30 FPS speed:

**cfg.enable_stream(RS2_STREAM_INFRARED, 1280, 720, RS2_FORMAT_BGR8, 30);**

blakat360, 2020-08-02:
I was just calling pipeline.start(). I've previously used a config object to enable individual cameras via cfg.enable_device(serialNumber);

the following throws an error:
```
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_INFRARED, 1280, 720, RS2_FORMAT_BGR8, 30);
rs2::pipe;
rs2::pipeline.start(cfg); // <---- error thrown on this line
```
 The console readout is:
```
terminate called after throwing an instance of 'rs2::error'
  what():  Couldn't resolve requests
```

I've googled 'rs2::config for using left IR stream as colour AND (D435 OR D400 OR D415)' and similar queries but found nothing useful.

MartyG-RealSense, 2020-08-03:
Please check whether the basic pipeline setup code below works for your script:

**rs2::pipeline pipe;
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_INFRARED, 1280, 720, RS2_FORMAT_BGR8, 30);
pipe.start(cfg);**

blakat360, 2020-08-03:
It does not

I ran the following:
```
#include <librealsense2/rs.hpp>

int main() {
  rs2::pipeline pipe;
  rs2::config cfg;
  cfg.enable_stream(RS2_STREAM_INFRARED, 1280, 720, RS2_FORMAT_RGB8, 30);
  pipe.start(cfg); //<----- error thrown here

  return 0;
}
```

Console readout is:
```
terminate called after throwing an instance of 'rs2::error'
  what():  Couldn't resolve requests
```

MartyG-RealSense, 2020-08-03:
Are you using the camera on a USB 2 connection instead of USB 3 please?  That may explain the error message, as a much more limited range of modes are available on USB 2 compared to USB 3 (as "Couldn't resolve requests" can mean that a resolution / FPS mode that was defined is not available).

Sometimes the camera may be plugged into a USB 3 port but is being mis-detected as USB 2.

blakat360, 2020-08-03:
I'm using a USB-C connection.

MartyG-RealSense, 2020-08-03:
If you are using your own choice of cable rather than the official short one supplied with the camera and the chosen cable is a USB 2 cable then that may have an outcome on how the camera is detected.

An easy way to tell whether your camera is being detected as USB 2 or USB 3 is to open the RealSense Viewer and look at the number beside the camera name at the top of the options side-panel.  If it says **2.1** beside the name then it is operating in USB 2 mode, which has a limited range of resolutions / FPS speeds available.  If it says **3.2** then it is a full-capability USB 3 connection.

blakat360, 2020-08-03:
It shows 3.2.

MartyG-RealSense, 2020-08-03:
Though the D435 has monochrome color in the left and right IR sensors, it seems that the RealSense SDK is only going to allow D415 models to access the IR imager's color mode.  This may be why it cannot support the requested stream definition.  I do apologise.

If we cannot use the IR color mode to deal with occlusion around the arm, the another way to tackle it may be to use the full HD color capabilities of the RGB sensor.

**cfg.enable_stream(RS2_STREAM_COLOR, 1920, 1080, RS2_FORMAT_BGR8, 30);**

blakat360, 2020-08-03:
This runs;

However, when I do the following it seems there is no depth frame in the frameset:
```
rs2::pipeline pipe;
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_COLOR, 1920, 1080, RS2_FORMAT_BGR8, 30);
pipe.start();

auto frameset = pipe.wait_for_frames()
rs2::colorizer().process(frameset.get_depth_frame()) // <--- process throws an error
```

console output is:
```
terminate called after throwing an instance of 'rs2::error'
  what():  null pointer passed for argument "frame"
```

To be clear, I only care about removing occlusion and am perfectly happy with dealing with a  monochrome image (even one which has an IR pattern on it) as this is not something the end client will see.

Are there any plans to add this functionality to the SDK and is there any way I can help with making it happen?

MartyG-RealSense, 2020-08-03:
There is no depth frame because one has not been defined in cfg statements.  When cfg custom configurations are used, it is necessary to tell the camera which streams you need.  So you should add this line below the color cfg line:

cfg.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16, 30);

Intel actually added an "occlusion invalidation" mechanism to the SDK's point-cloud calculations from version 2.35 onward.  They discuss it in their recent white-paper document on projection and occlusion.

https://dev.intelrealsense.com/docs/projection-texture-mapping-and-occlusion-with-intel-realsense-depth-cameras#section-4-occlusion-invalidation

Would it be possible to provide an image that shows the occlusion of the arm, please?  A "good" occlusion should surround an object relatively equally rather than be heavily biased towards one side of the object (e.g all on the right side of the arm).

blakat360, 2020-08-03:
I attach a comparison.

[image]
[image]

It seems that with the custom config the general quality of the depth data is reduced (i see more holes).

Are there workarounds which use multiple cameras? Would I get significantly  better results with a D415?

MartyG-RealSense, 2020-08-03:
The occlusion is definitely biased towards one side of the arm.  It may be worth performing a **Dynamic Calibration** of the camera to ensure that it is not misaligned.

As you are using Ubuntu, you can install the Linux version of the tool using the instructions on page 14 onward of the tool's user guide:

https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

The new RealSense D455 contains improvements designed to improve depth to color alignment.  Both RGB and depth imagers are mounted on the same "stiffener" inside the camera, and both imagers have the same field of view (FOV) size.  RGB FOV is smaller than depth FOV on the D435 models (and the same FOV size on D415).

D415 has around 2x better image quality than the D435 models.

dorodnic, 2020-08-03:
Hi @blakat360 
Not sure I follow. Occlusion is symmetric, its points that one sensor sees and the other one does not. 
It may be possible to align depth to Infrared 2 but this would not reveal any new depth information. Ultimately, if there is an object obstructing one of the IR sensors, it's impossible to do stereo matching in that area. 

What @MartyG-RealSense is suggesting is reducing occlusion between _depth and RGB_ sensor by using colour from left IR. There is no way to reduce occlusion between left and right IR, at least that I know of.

MartyG-RealSense, 2020-08-03:
Thanks very much @dorodnic for your advice!  My understanding is that @blakat360 basically wants to remove the apparent occlusion.   I was working under the impression that @blakat360 was using a point cloud.  I guess that if it is a 2D image then it will not benefit from the occlusion invalidation algorithm?

dorodnic, 2020-08-03:
Every pair of cameras has its own area of occlusion. When using point-cloud with texture mapping to RGB (or aligned streams) there is occlusion between depth and RGB sensors. In addition, there is always occlusion between left and right, appearing as black outlines around objects, black band on the left, and other artefacts of similar nature.

closed 2020-08-16 by MartyG-RealSense