realsenseai/librealsense #7001: The function of "context.query_devices()" can't List all devices

zhuhe1113, 2020-08-03:
I have two realsense D435, I want to achieve hardware synchronization. The first step is to list all devices through function "cxt.query_devices()", but  the function "query_devices()" return the same device.
  this  was my Code:

for (auto && dev : ctx.query_devices())
{
    cout << dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER) << endl;
}

the Code  outputs two identical serial numbers:
814412072461
814412072461

but When I connect one of the devices separately, two different serial numbers are output:
814412072461
740112070722

MartyG-RealSense, 2020-08-03:
Hi @zhuhe1113   I hope that the discussion in the link below will be helpful to you.

https://github.com/IntelRealSense/librealsense/issues/2219

It is a long conversation but very educational and all of it is worth reading.  For a quick jump to a scripting reference though, this script may be what you need:

https://github.com/IntelRealSense/librealsense/issues/2219#issuecomment-412350887

zhuhe1113, 2020-08-03:
> Hi @zhuhe1113 I hope that the discussion in the link below will be helpful to you.
> 
> #2219
> 
> It is a long conversation but very educational and all of it is worth reading. For a quick jump to a scripting reference though, this script may be what you need:
> 
> [#2219 (comment)](https://github.com/IntelRealSense/librealsense/issues/2219#issuecomment-412350887)

Thank you for your help, I have solved this problem

MartyG-RealSense, 2020-08-03:
Great news that you achieved a solution - thanks for the update!   Is it okay to close this case now if you are satisfied with the outcome, please?

closed 2020-08-15 by MartyG-RealSense