realsenseai/librealsense #7002: T265 VIO source code

oym1994, 2020-08-03:
Thanks for your great effort! 
Now I am researching on VIO , very interested in the VIO built in T265 camera. I wander how can I get the source code.(I can't find it in the github). Thank you very much!

dorodnic, 2020-08-03:
Hello @oym1994 
At the moment the algorithm is a proprietary and not available as open-source.

closed 2020-08-05 by dorodnic

arjunsengupta98, 2020-09-21:
Hello @dorodnic 
Is there any plan of making the algorithm open-source in the future?

Since the team does not plan on addressing issues related to T265 in the future (as mentioned in  https://github.com/IntelRealSense/librealsense/issues/7258#issuecomment-692964955), I was wondering if there are any such plans of disclosing your algorithm. 

I'm not exactly being able to accomplish my purpose with the T265, and was thinking about running VIO algorithms on my development board itself.