realsenseai/librealsense #7003: How to filter strange black spots in my Pointcloud from L515 ?

zainmehdi, 2020-08-03:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 18.01 | 
| Platform                                 | PC|
| SDK Version                            |  2.35   | 
| Segment			| Realsense Viewer           | 

### Issue Description
I am saving ply using the provided save_to_ply function in C++. But in my pointcloud I am getting black pixels. I dont get them while I use real-sense viewer to save my data. Can someone please guide me in the right direction I am not sure what am I doing wrong. Thanks

[image]

Here is the code snippet
`      for (auto &&pipe : pipelines)
        {
            rs2::frameset frames;

            if (pipe.poll_for_frames(&frames))
            {

                
                frames=align_to_depth.process(frames);
                // getting the color frame;
                auto color = frames.get_color_frame();

                // For cameras that dont have RGB we will map it to IR space
                if (!color)
                {
                    auto colot = frames.get_infrared_frame();
                }

                // Mapping pointcloud to this color space
                if (color)
                {
                    pc.map_to(color);
                    save_frame_color_data("/home/zain/data_atlas3D/rgb", color);
                }

                // Getting depth data
                auto depth = frames.get_depth_frame();

                if (depth)
                {
                    
                    //depth = dec_filter.process(depth);
                    depth = thr_filter.process(depth);
                    //depth = spat_filter.process(depth);
                    //depth = temp_filter.process(depth);
                    depth=hole_filter.process(depth);

                    pc.calculate(depth);
                    
                  
                    points = pc.calculate(depth);
                    save_frame_pointcloud(points,"/home/zain/data_atlas3D/depth",color);

                    save_frame_depth_data("/home/zain/data_atlas3D/depth", depth, depth_units);
                }

                // Getting Pose data
                auto pose = frames.get_pose_frame();
                if (pose)
                {
                    save_frame_pose_data("/home/zain/data_atlas3D/pose", pose);
                }
            }
        }`

vigneshsara, 2020-08-03:
hi ,
i think your problem on filters , can you try this program without adding the thr_filter and hole_filters.

zainmehdi, 2020-08-03:
@vigneshsara Thanks. I will try

MartyG-RealSense, 2020-08-04:
@zainmehdi   I am not able to handle L515 related cases as they are the responsibility of someone else on the RealSense team.

zainmehdi, 2020-08-04:
@MartyG-RealSense Thank you for your response. @RealSenseCustomerSupport @dorodnic Can I please get some help in my case. I tried the hole filling filter and threshold filter as well with no considerable improvement. Will there be any white paper or specific instructions regarding tuning of l515 ? 
Thanks

RealSenseSupport, 2020-08-14:
Hi @zainmehdi 

There are some tuning information that we have published.  Here are some links below:

https://support.intelrealsense.com/hc/en-us/articles/360051646094-Intel-RealSense-LiDAR-Camera-L515-User-Guide

https://www.intelrealsense.com/optimizing-the-lidar-camera-l515-range/

https://dev.intelrealsense.com/docs/projection-texture-mapping-and-occlusion-with-intel-realsense-depth-cameras

I wonder if there is occlusion happening that is causing what you're seeing.

RealSenseSupport, 2020-09-11:
@zainmehdi 

Is there anything further needed on this one?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-09-14 by zainmehdi