realsenseai/librealsense #7006: why my point cloud is merge to background data .

vigneshsara, 2020-08-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Linux (Ubuntu 18.0   |                                       
| Platform                                 | PC  |
| SDK Version                            |   2                           | 
| Language                            |  C++                          | 
| Segment			|  Robot                           | 

### Issue Description
i have taken the point cloud data in high_density mode in 1000mm and 750mm height, on 1000mm my point cloud data is merge to background(ground) , if i decrease the height to 750mm its not look like that , please ref the below images
 
750mm height image 
[image]

1000mm height image 
[image]

1. this is except behavior in depth algorithm when increase the height ???
2. Its is any possible to get the 750mm  behavior in 1000mm height.

MartyG-RealSense, 2020-08-03:
Hi @vigneshsara   The 1000 mm point cloud looks much more "wavy" compared to the 750 mm distance.  Increased waviness in the point cloud can be a known consequence as the camera is moved further away (if increasing the distance of the camera from the observed object is what you mean by "height").

There are a couple of very good references about point cloud waviness:

https://github.com/IntelRealSense/librealsense/issues/1375
https://github.com/IntelRealSense/librealsense/issues/4553

Could you test whether you get an improved cloud with the "Default" preset please, as this has clean edges and reduced point cloud spraying.

vigneshsara, 2020-08-03:
HI @MartyG-RealSense  thank for your replay . but my question is not based on waviness nature of the cloud. Its is split between the object and  background (ground) .in 750mm the split is more visible but in 1000mm split is not visible.

MartyG-RealSense, 2020-08-03:
Error in depth measurement scales linearly with distance.  The further that the camera moves away from the observed object, the greater the amount of error in the measurements becomes.  This is known as **RMS error**.  

It looks to me though that the separation in the 750 mm image is not a "split" but an area that is empty of depth data because the camera cannot read depth detail from that side of the object for some reason.  Such areas are typically rendered as plain black, as there is no depth detail in them.  

In the 1000 mm image, you can see the edges of the box much better.  So in that respect, the 1000 mm image is the better depth image of the two (aside from the increased waviness).

closed 2020-08-27 by MartyG-RealSense