realsenseai/librealsense #7010: Open infrared & depth stream with rs2::syncer

tom-garford, 2020-08-03:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 2.36 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            |  2.36}                          | 
| Language                            |  C++|                          | 

### Issue Description

Simple question, although I can't find an example anywhere: how do I open an infrared stream (Infrared sensor #1) and depth stream on the stereo module at the same time using the rs2::syncer class. 

This is obviously easy with pipe, config.enable_stream, etc., etc. although I am keen to use the syncer class instead. 

doing `depth_sensor.open(sp_ir)` and `depth_sensor.open(sp_dep)` results in this error: 

`open(...) failed. UVC device is already opened!`

fairly unsurprisingly. As the stream profiles for IR and depth both use stereo module numbering this figures although it is obviously possible. 

Is rs2::syncer able to be used with IR and depth streams as they are by definition synced? 

Please let me know.

Tom.

dorodnic, 2020-08-03:
Hi @tom-garford 
Please try something like `depth_sensor.open({sp_dep, sp_ir})`, the overload taking vector of stream profiles. The camera requires IR and depth to be opened at the same time.

tom-garford, 2020-08-04:
Hi @dorodnic 

Thanks very much for your swift answer. That works great, as does 
`std::vector<rs2::stream_profile> 
sp_dep_ir = depth_sensor.get_stream_profiles()  -->  depth_sensor.open(sp_dep_ir) ` 
when opening a rosbag with a depth and IR stream. 

Thanks very much. 

Tom.

closed 2020-08-04 by tom-garford