realsenseai/librealsense #7012: L515 Recording - Failed to write stream extriniscs

JeremyBYU, 2020-08-03:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.04.01.02 | 
| Operating System & Version |   Windows and Linux x64 | 
| Kernel Version (Linux Only)    |  5.3                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.36                           | 
| Language                            |  python                          | 
| Segment			|  Robot                        | 

### Issue Description

When recording the L515 Depth and RGB stream using the RealSense Viewer the extrinsics (rotation/translation between rbg sensor to depth sensor) are not recorded properly into the `bag` file. All other data seem to be properly recorded and there are no issues with the depth or RGB streams. This only happens with the L515 sensor. My D435i records everything perfectly. There are three pieces of evidence for this. 

1. The `stdout` of `realsense-viewer` prints an error message -  `03/08 12:41:20,857 WARNING [139937040922368] (ros_writer.cpp:168) Failed to write stream extrinsics for 2. Exception: map::at`
2. When using the `rs-rosbag-inspector` you can see that the `tf` *topic* is missing. This is the ROS formatted type that contains the `transform` or extrinsic. You can see it missing in the first picture below. The second picture shows my D435i having the correct topic. 

Bag File Inspector - L515 - No Extrinsic Message 

[image]

Bag File Inspector - 435i - Extrinsic Message available

[image]

3. When using the `align` post processing on the recorded L515 bag file (during [playback](https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.playback.html)) the color and depth streams do not become aligned. I believe the absence of the extrinsic treats it as the identify matrix (no translation or rotation). When **not** using the bag file, but instead livestreaming, the alignment is nearly perfect. 

Suggested actions:

Follow the error in the log file that is attached. My guess is maybe an implicit assumption that the rgb should be aligned to an **infrared** camera 0 whereas the L515 does not have one. This is just a wild guess. I will try and take a look as well into `ros_writer.cpp`.

```
(ppb) ➜  polylidar-realsense git:(l515) realsense-viewer
 03/08 12:41:14,013 WARNING [139937040922368] (types.cpp:78) create_snapshot(...) not implemented!
 03/08 12:41:14,014 ERROR [139937040922368] (record_device.cpp:217) Failed to add snapshot of type librealsense::debug_interface. Exception: create_snapshot(...) not implemented!
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Motion
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Video
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Roi
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Depth Frame
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Advanced Mode
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Video Profile
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Disparity Frame
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Motion Profile
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Pose Frame
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Pose Profile
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Tm2
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Software Device
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Software Sensor
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Decimation Filter
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Threshold Filter
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Disparity Filter
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Spatial Filter
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Temporal Filter
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Hole Filling Filter
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Zero Order Filter
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Pose
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Pose Sensor
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Wheel Odometer
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Global Timer
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Updatable
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Update Device
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Tm2 Sensor
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Auto Calibrated Device
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Depth Huffman Decoder
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Serializable
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Fw Logger
 03/08 12:41:14,014 WARNING [139937040922368] (record_device.cpp:260) Extensions type is unhandled: Auto Calibration Filter
[... 202 lines cut ...]
 03/08 12:41:20,289 WARNING [139937040922368] (ros_writer.cpp:168) Failed to write stream extrinsics for 2. Exception: map::at
 03/08 12:41:20,353 WARNING [139937040922368] (ros_writer.cpp:168) Failed to write stream extrinsics for 2. Exception: map::at
 03/08 12:41:20,415 WARNING [139937040922368] (ros_writer.cpp:168) Failed to write stream extrinsics for 2. Exception: map::at
 03/08 12:41:20,477 WARNING [139937040922368] (ros_writer.cpp:168) Failed to write stream extrinsics for 2. Exception: map::at
 03/08 12:41:20,540 WARNING [139937040922368] (ros_writer.cpp:168) Failed to write stream extrinsics for 2. Exception: map::at
 03/08 12:41:20,601 WARNING [139937040922368] (ros_writer.cpp:168) Failed to write stream extrinsics for 2. Exception: map::at
 03/08 12:41:20,667 WARNING [139937040922368] (ros_writer.cpp:168) Failed to write stream extrinsics for 2. Exception: map::at
 03/08 12:41:20,730 WARNING [139937040922368] (ros_writer.cpp:168) Failed to write stream extrinsics for 2. Exception: map::at
 03/08 12:41:20,795 WARNING [139937040922368] (ros_writer.cpp:168) Failed to write stream extrinsics for 2. Exception: map::at
 03/08 12:41:20,857 WARNING [139937040922368] (ros_writer.cpp:168) Failed to write stream extrinsics for 2. Exception: map::at
 03/08 12:41:20,921 WARNING [139937040922368] (ros_writer.cpp:168) Failed to write stream extrinsics for 2. Exception: map::at
 03/08 12:41:20,982 WARNING [139937040922368] (ros_writer.cpp:168) Failed to write stream extrinsics for 2. Exception: map::at
 03/08 12:41:21,046 WARNING [139937040922368] (ros_writer.cpp:168) Failed to write stream extrinsics for 2. Exception: map::at
 03/08 12:41:21,108 WARNING [139937040922368] (ros_writer.cpp:168) Failed to write stream extrinsics for 2. Exception: map::at
 03/08 12:41:21,171 WARNING [139937040922368] (ros_writer.cpp:168) Failed to write stream extrinsics for 2. Exception: map::at
[... cut ...]
 03/08 12:41:21,426 WARNING [139937040922368] (ros_writer.cpp:168) Failed to write stream extrinsics for 2. Exception: map::at
 03/08 12:42:10,300 WARNING [139937517497280] (types.cpp:78) This option is read-only!
 03/08 12:42:10,311 WARNING [139937040922368] (playback_device.cpp:198) Playback device does not provide a matcher
 03/08 12:42:10,315 WARNING [139937032529664] (playback_device.cpp:198) Playback device does not provide a matcher
 03/08 12:42:10,340 WARNING [139936544061184] (playback_device.cpp:198) Playback device does not provide a matcher

```

JeremyBYU, 2020-08-04:
Found the issue. There is a bug inside the L515 extrinsics code contained in `src/l500/l500-color.cpp`. I believe this is simply a copy-paste error when creating the code for the L515 color sensor from the depth sensor. 

Briefly here is the git diff to fix the issue:

```
(base) jeremy@jeremy-VirtualBox:~/Documents/librealsense$ git diff
diff --git a/src/l500/l500-color.cpp b/src/l500/l500-color.cpp
index ac0202ce9..a6baadda8 100644
--- a/src/l500/l500-color.cpp
+++ b/src/l500/l500-color.cpp
@@ -164,7 +164,8 @@ namespace librealsense
                 return get_color_stream_extrinsic(*_color_extrinsics_table_raw);
             } );
         environment::get_instance().get_extrinsics_graph().register_extrinsics(*_depth_stream, *_color_stream, _color_extrinsic);
-        register_stream_to_extrinsic_group(*_depth_stream, 0);
+        // register_stream_to_extrinsic_group(*_depth_stream, 0);
+        register_stream_to_extrinsic_group(*_color_stream, 0);
 
 
         _color_device_idx = add_sensor(create_color_device(ctx, color_devs_info));
```

This git diff comes from the release tagged at `2.36`. The issue is still present on the master branch:

https://github.com/IntelRealSense/librealsense/blob/f7cdf6e8961e1709f6d864bdb33095c00a671ca7/src/l500/l500-color.cpp#L167

We should be registering the *color* stream for the extrinsis, not the depth. I hunted this bug down through the following steps:

1. Set up Ubuntu 18.04 virtual machine with USB 3.0 pass through using virtual box. Install all real-sense dependencies and build realsense in debug.
2. Put breakpoints inside the function that was throwing an error from the  log file in the OP:

https://github.com/IntelRealSense/librealsense/blob/f7cdf6e8961e1709f6d864bdb33095c00a671ca7/src/media/ros/ros_writer.cpp#L117-L126
3. Determined the error is throwing in `dev.get_extriniscs`
4. Determined was throwing in the function `.at(..)`
https://github.com/IntelRealSense/librealsense/blob/f7cdf6e8961e1709f6d864bdb33095c00a671ca7/src/device.cpp#L233-L236
5. Realized that the unique stream id of `3` was **not** a `key` inside of `_extrinsics`.

At this point I hunted around to find the code of when _extrinsics `map` is modified. This lead me `l500-color.cpp` where this line of code will modify `_extrinsics`. I noticed that is was passing and registering the **depth** stream, even though what we wanted was the color:

https://github.com/IntelRealSense/librealsense/blob/f7cdf6e8961e1709f6d864bdb33095c00a671ca7/src/l500/l500-color.cpp#L167

I looked at the sister of L515, D4XX code to see what it does:

https://github.com/IntelRealSense/librealsense/blob/f7cdf6e8961e1709f6d864bdb33095c00a671ca7/src/ds5/ds5-color.cpp#L52

It registers the color stream. I therefore modified this code (L500) and verified it working with the following tests:

1. No error messages in log file
2. Save bag file has the correct `tf` topic for the color sensor
3. My code that reads the bag file can now properly align the streams

I would appreciate it if an Intel engineer can verify that this is the correct fix. If so you can merge this simple change or I can make a pull request if desired.

 

### Appendix - Note

I also attempted to change this code to match the DS5 as a test.

https://github.com/IntelRealSense/librealsense/blob/f7cdf6e8961e1709f6d864bdb33095c00a671ca7/src/l500/l500-color.cpp#L166
https://github.com/IntelRealSense/librealsense/blob/f7cdf6e8961e1709f6d864bdb33095c00a671ca7/src/ds5/ds5-color.cpp#L51

This **did not** work and made the alignment of the color and depth streams in the viewer broken. The alignment in the viewer was already working (i believe it is using the extrinsics graph this line is referring to) so I didn't think this was necessary, however I changed it to see what would happen.

ev-mp, 2020-08-04:
@JeremyBYU , thank you for the in-depth analysis provided - it is concise and clear.
I think though that in the meanwhile it has been already fixes internally via https://github.com/IntelRealSense/librealsense/pull/6954 and merged into `development` branch.

Can you try to build from `development` and check whether would rectify the issue for you?

JeremyBYU, 2020-08-04:
@ev-mp, Yep you are absolutely correct that #6954 fixes the problem. I searched for this problem in the the "Issues" tab but should have also searched in "Pull Requests". I also should have checked the development branch and not just the master.  Sorry about the noise! At least there will be more connections between the "cure" in #6954 and the symptoms reported in this issue.

I pulled down the development branch and verified (no log warnings during recording, rs-bag-inspector, user `align` code) that all is working as expected.

closed 2020-08-04 by JeremyBYU