realsenseai/librealsense #7013: Unreal plugin

jeroenvangeel, 2020-08-03:
| Camera Model                       | D435i } | 
| Firmware Version                   | 05.12.06.00 | 
| Operating System & Version | Win 10 | 
| SDK Version                            | WIN10-2.36.0.2034 | 
| Segment			|  Unreal | 

### Issue Description
 I tried getting the unreal engine plugin installed but got the following error when i try to build te projectfile:
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/unrealengine4

(those are in my native language, il translated them)
`error CS1729: UnrealBuildTool.ModuleRules does not contain a constructor that takes 0 arguments`
` error CS0103: The name UEBuildConfiguration does not exist in the current context`
`ERROR: Unable to compile source files.`

Think the document is a bit outdated maybe?
I followed some links to a video showing to add the plugin, also this didnt work for me.
seems i download a different plugin as seen in the video.
https://software.intel.com/content/www/us/en/develop/videos/ue4-adding-the-intel-realsense-plugin-to-a-new-project.html

MartyG-RealSense, 2020-08-03:
There was a past UE4 plugin for the 2015 / 2016 version of the RealSense SDK for Windows that is now obsolete and not updated.  This SDK is not compatible with modern RealSense cameras such as the 400 Series.

https://github.com/GameTechDev/UE4RealSensePlugin

For the 400 Series cameras, the Unreal Engine 4 wrapper in your first link should be used.  You do not need to install the old 2015 / 2016 plugin.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/unrealengine4

Some users of the UE4 wrapper have found that a minimum of UE 4.23.1 should be used now.

https://github.com/IntelRealSense/librealsense/issues/5359

jeroenvangeel, 2020-08-03:
was on unreal 4.8.
it could actually be i used the old plugin.
works on 4.25.3 now!

MartyG-RealSense, 2020-08-03:
Excellent news - thanks for the update!

MartyG-RealSense, 2020-08-16:
Closed due to no further comments received.

closed 2020-08-16 by MartyG-RealSense