realsenseai/librealsense #7019: L515 Android Connectivity Problems on Samsung Galaxy Tab S4 (SM-T830)

rafaelspring, 2020-08-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.04.01.00 | 
| Operating System & Version |  Android 10 | 
| Kernel version | 4.4.205-18876299 |
| SDK Version                            |  2.36.0  | 
| Language                            |  C++ / Java | 

### Issue Description
Error happened while streaming data in this configuration:
Depth at 640x480 at 30 Hz
RGB at 1280x720 at 30 Hz
IMU at 200 Hz

After a while the sensor was no longer streaming data and we got the following error logs.
The issue is repeatable.

```
Data streaming fine, then:
...
E librs   : uvc streamer watchdog triggered on endpoint: 131
E librs   : uvc streamer watchdog triggered on endpoint: 133
E librs   : uvc streamer watchdog triggered on endpoint: 130
...
W librs   : control_transfer returned error, index: 768, error: Connection timed out, number: 110
W librs   : control_transfer returned error, index: 768, error: Cannot send after transport endpoint shutdown, number: 108
W librs   : control_transfer returned error, index: 130, error: Connection timed out, number: 110
W librs   : Failed to reset the USB pipe 130, error: RS2_USB_STATUS_TIMEOUT
W librs   : control_transfer returned error, index: 133, error: Connection timed out, number: 110
W librs   : Failed to reset the USB pipe 133, error: RS2_USB_STATUS_TIMEOUT
W librs   : control_transfer returned error, index: 131, error: Connection timed out, number: 110
W librs   : Failed to reset the USB pipe 131, error: RS2_USB_STATUS_TIMEOUT
W librs   : usb_request_queue returned error, endpoint: 139 error: No such device, number: 19
W librs   : usb_request_queue returned error, endpoint: 131 error: No such device, number: 19
E librs   : failed to submit UVC request
E librs   : failed to submit UVC request, error: -4
W librs   : usb_request_queue returned error, endpoint: 131 error: No such device, number: 19
E librs   : failed to submit UVC request, error: -4
W librs   : usb_request_queue returned error, endpoint: 133 error: No such device, number: 19
E librs   : failed to submit UVC request, error: -4
W librs   : usb_request_queue returned error, endpoint: 133 error: No such device, number: 19
E librs   : failed to submit UVC request, error: -4
W librs   : usb_request_queue returned error, endpoint: 130 error: No such device, number: 19
E librs   : failed to submit UVC request, error: -4
W librs   : usb_request_queue returned error, endpoint: 130 error: No such device, number: 19
E librs   : failed to submit UVC request, error: -4
W librs   : usb_request_queue returned error, endpoint: 136 error: No such device, number: 19
...
I librs Enumerator: onReceive: android.hardware.usb.action.USB_DEVICE_DETACHED
I librs MessagesHandler: handleMessage: realsense device detached
I librs Enumerator: notifyOnDetach
D librs DeviceWatcher: Removing device: /dev/bus/usb/002/002
...
D librs DeviceWatcher: Device: /dev/bus/usb/002/002 removed successfully
W librs   : control_transfer returned error, index: 768, error: Bad file descriptor, number: 9
W librs   : control_transfer returned error, index: 768, error: Bad file descriptor, number: 9
E librs   : Error during time_diff_keeper polling: usb device disconnected
E librs   : Error during polling error handler: usb device disconnected
W librs   : usb_request_cancel returned error, endpoint: 133, error: Bad file descriptor, number: 9
W librs   : usb_request_cancel returned error, endpoint: 133, error: Bad file descriptor, number: 9
W librs   : control_transfer returned error, index: 133, error: Bad file descriptor, number: 9
W librs   : Failed to reset the USB pipe 133, error: RS2_USB_STATUS_NO_DEVICE
W librs   : usb_request_cancel returned error, endpoint: 139, error: Not a typewriter, number: 25
W librs   : control_transfer returned error, index: 7, error: Bad file descriptor, number: 9
W librs   : control_transfer of USB_REQUEST_CODE_GET failed return value is: -4
W librs   : control_transfer returned error, index: 7, error: Bad file descriptor, number: 9
W librs   : control_transfer of USB_REQUEST_CODE_GET failed return value is: -4
...
I librs   : Found 0 RealSense devices (mask 254)
...
I librs Enumerator: onReceive: android.hardware.usb.action.USB_DEVICE_ATTACHED
D librs UsbUtilities: hasUsbPermission
D librs UsbUtilities: grantUsbPermissions
I librs UsbUtilities: grantUsbPermissions:
I librs UsbUtilities: device: UsbDevice[mName=/dev/bus/usb/002/002,mVendorId=32902,mProductId=2916,mClass=239,mSubclass=2,mProtocol=1,mManufacturerName=Intel(R) RealSense(TM) Camera,mProductName=Intel(R) RealSense(TM) 515,mVersion=10.41,mSerialNumberReader=android.hardware.usb.IUsbSerialReader$Stub$Proxy@56b892f,mConfigurations=[
I librs UsbUtilities: UsbConfiguration[mId=1,mName=null,mAttributes=192,mMaxPower=55,mInterfaces=[
I librs UsbUtilities: UsbInterface[mId=0,mAlternateSetting=0,mName=Intel(R) RealSense(TM) 515 Depth,mClass=14,mSubclass=1,mProtocol=0,mEndpoints=[
I librs UsbUtilities: UsbEndpoint[mAddress=136,mAttributes=3,mMaxPacketSize=64,mInterval=10]]
I librs UsbUtilities: UsbInterface[mId=1,mAlternateSetting=0,mName=Intel(R) RealSense(TM) Camera I,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
I librs UsbUtilities: UsbEndpoint[mAddress=131,mAttributes=2,mMaxPacketSize=1024,mInterval=0]]
I librs UsbUtilities: UsbInterface[mId=2,mAlternateSetting=0,mName=Intel(R) RealSense(TM) 515 Depth,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
I librs UsbUtilities: UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=1024,mInterval=0]]
I librs UsbUtilities: UsbInterface[mId=3,mAlternateSetting=0,mName=Intel(R) RealSense(TM) Camera C,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
I librs UsbUtilities: UsbEndpoint[mAddress=132,mAttributes=2,mMaxPacketSize=1024,mInterval=0]]
I librs UsbUtilities: UsbInterface[mId=4,mAlternateSetting=0,mName=Intel(R) RealSense(TM) 515 RGB,mClass=14,mSubclass=1,mProtocol=0,mEndpoints=[]
I librs UsbUtilities: UsbInterface[mId=5,mAlternateSetting=0,mName=Intel(R) RealSense(TM) 515 RGB,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
I librs UsbUtilities: UsbEndpoint[mAddress=133,mAttributes=2,mMaxPacketSize=1024,mInterval=0]]
I librs UsbUtilities: UsbInterface[mId=6,mAlternateSetting=0,mName=null,mClass=255,mSubclass=0,mProtocol=0,mEndpoints=[
I librs UsbUtilities: UsbEndpoint[mAddress=1,mAttributes=2,mMaxPacketSize=1024,mInterval=0]
I librs UsbUtilities: UsbEndpoint[mAddress=129,mAttributes=2,mMaxPacketSize=1024,mInterval=0]]
I librs UsbUtilities: UsbInterface[mId=7,mAlternateSetting=0,mName=Intel(R) RealSense(TM) HID,mClass=3,mSubclass=0,mProtocol=0,mEndpoints=[
I librs UsbUtilities: UsbEndpoint[mAddress=139,mAttributes=3,mMaxPacketSize=64,mInterval=2]]]]
I librs MessagesHandler: handleMessage: realsense device attached
I librs Enumerator: notifyOnAttach
```

dorodnic, 2020-08-04:
Hi @rafaelspring 
Some work towards Android stability with L515 was done here - #6945 
We are not done and currently investigating stability issues on RK3399, but the pull-requests should improve general stability

rafaelspring, 2020-08-04:
@dorodnic Thanks for the quick update. I will try this out.
The device in question is this one: https://www.gsmarena.com/samsung_galaxy_tab_s4_10_5-9262.php
It uses a Snapdragon 835.

rafaelspring, 2020-08-06:
@dorodnic I have tried #6945 as well as well as e3c0b49 and predecessors on @gwen2018 's fork.
Using a libRS build with these changes unfortunately after about 10 seconds of operation I no longer receive new RGB frames, i.e. the stream just stops delivering.

gwen2018, 2020-08-09:
@rafaelspring, thanks for the feedback. can you please attach a more complete log file, especially with the part before the wacthdog and errors start to appear, including logs from kernel, logcat V -b all? Also, we have a new branch below with experimental fixes for another similar issue, I have tried on a different platform, the same stream configuration as yours worked fine. If not, please try use smaller 640x480 RGB resolution as well.

 https://github.com/gwen2018/librealsense/tree/rs_android_lost_cam_fix

Please let us know how it goes. Thanks in advance for trying.

RealSenseSupport, 2020-08-19:
Hi @rafaelspring ,

Have you had an opportunity to try the newer branch @gwen2018 provided?

Thanks

RealSenseSupport, 2020-09-01:
Hi,

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2020-09-08 by RealSenseSupport

Shanki070, 2021-08-23:
While detaching the USB cable getting the same issue. Is this issue fixed?