realsenseai/librealsense #7020: rs2_transform_point_to_point

LorenzoMonti, 2020-08-04:
Hello,

I'm working with 3 Intel D415 and especially with depth streams. I would need to align some 3D points, taking a master camera as a reference point (as coordinate space). To do this, I applied the _rs2_transform_point_to_point_ function taking the extrinsic data from the master camera and the list of 3D points of the workers cameras. The function, however, returns the starting points of the other cameras (in practice it seems that it does not apply anything).

any suggestions?

Thanks in advance

p.s. this is the rotation and translation matrix: rotation: [1, 0, 0, 0, 1, 0, 0, 0, 1], translation: [0, 0, 0]

MartyG-RealSense, 2020-08-04:
Hi @LorenzoMonti    Would using ROS be an option for you?   If so, Intel has RealSense ROS tutorials on combining point clouds from multiple cameras, using 1 or 2 machines:

**1 machine**
In this tutorial, you can put all the cameras on one computer.  The tutorial starts with a 2-camera setup but later explains how to add further cameras.

https://github.com/IntelRealSense/realsense-ros/wiki/Showcase-of-using-2-cameras

**2 machines**
In this tutorial, 3 cameras are spread across 2 computers.

https://github.com/IntelRealSense/realsense-ros/wiki/showcase-of-using-3-cameras-in-2-machines

MartyG-RealSense, 2020-08-16:
Hi @LorenzoMonti   Due you require further assistance with this case please, or can it be closed?  Thanks!

closed 2020-08-27 by MartyG-RealSense