realsenseai/librealsense #7024: Depth to IR alignment causes freeze.

rajrajeswaran, 2020-08-04:
Camera Model : D415
Firmware Version: 05.12.06.00
Operating System & Version: Windows 10 Enterprise (v1909)
Platform: PC
SDK Version: 2.36.0.2034
Language: C#                            

I am having an issue when aligning depth to IR data.  When i run the following program, after few seconds the feed will freeze.   
It is constantly stopping at 21st frame.  If i don't do the alignment, the feed will continue without any interruption. 

	    Align alignIr = new Align(Stream.Infrared);
            var cfg = new Config();
            cfg.EnableStream(Stream.Color);
            cfg.EnableStream(Stream.Depth);
            cfg.EnableStream(Stream.Depth, 640, 480);
            cfg.EnableStream(Stream.Color, 640, 480, Format.Bgra8, 15);
            cfg.EnableStream(Stream.Infrared, 1, 640, 480, Format.Y8, 15);
            cfg.EnableStream(Stream.Infrared, 2, 640, 480, Format.Y8, 15);
            var pipe = new Pipeline();
            pipe.Start(cfg);

            _keepRunning = true;
            while (_keepRunning)
            {
                try
                {
                    using (var frames = pipe.WaitForFrames(5000))
                    {
                        var irFrame = frames.InfraredFrame.DisposeWith(frames);
                        var colorFrame = frames.ColorFrame.DisposeWith(frames);
                        var depth = frames.DepthFrame.DisposeWith(frames);

                        lblFrameInfo.Text = "Frame Info: " + irFrame.Number + "  " + irFrame.Timestamp + " ";
                        //if i remove the following line, the feed will continue w/o interruption.
                        var alignedFrame = irFrame.ApplyFilter(alignIr);
                        Application.DoEvents();
                    }
                }
                catch (Exception ex)
                {
                    MessageBox.Show("Error: " + ex.Message);
                }
            }

MartyG-RealSense, 2020-08-04:
Hi @rajrajeswaran    There are very few resources about aligning depth to infrared.  This is likely because - unlike RGB color - the left infrared imager is already perfectly pixel-aligned, calibrated and overlaid with the depth map by default.  

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#section-use-the-left-color-camera

You can align depth with any other stream type.  In the case of depth to infrared alignment though, the two are already aligned.

dorodnic, 2020-08-05:
Hi @rajrajeswaran 
Note that alignedFrame is not being deterministically disposed.

rajrajeswaran, 2020-08-05:
> Hi @rajrajeswaran
> Note that alignedFrame is not being deterministically disposed.

Thank you @dorodnic for pointing that out.  That does solve the problem.  Thanks

rajrajeswaran, 2020-08-05:
> Hi @rajrajeswaran There are very few resources about aligning depth to infrared. This is likely because - unlike RGB color - the left infrared imager is already perfectly pixel-aligned, calibrated [...]

@MartyG-RealSense - Thank you for the explanation.  Appreciate the link!

MartyG-RealSense, 2020-08-05:
You are very welcome, @rajrajeswaran   :)

rajrajeswaran, 2020-08-05:
This issue has been resolved!  See comments above.

closed 2020-08-05 by rajrajeswaran