realsenseai/librealsense #7029: Question about the confidence map of L515

cdb0y511, 2020-08-04:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (1.04.02) | 
| Operating System & Version |   (Ubuntu 18.04)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.36.0 }                          | 
| Language                            |  {c++}                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

Hi, I notice that the confidence map of L515 can be very useful. Since I set the confidence to 3.0 almost all bad measurements are gone, however, there are only three options in the viewer settings(1.0,2.0,3,0). I may use the original confidence map to enhance the final depth estimation, so there are the following questions:

1. What is the physical meaning of the confidence map? Signal noise ratio of the lidar? How does the confidence map calculate?

2.Is there any example in C++ to export the confidence map? Any example to utilize the confidence will be nice.

Thanks

cdb0y511, 2020-08-05:
It may related with:
https://github.com/IntelRealSense/librealsense/blob/8594d09f092347a8b3d832d14e4fb631140620c5/src/proc/zero-order.cpp
and
https://github.com/IntelRealSense/librealsense/issues/6775#issuecomment-660647667

cdb0y511, 2020-08-06:
I figure out how to read the confidence.
It is raw 8 bit value.
I suppose higher value means better depth estimation, right?
But what is the  physical meaning of the confidence map?
I really import for me.
Thanks

ev-mp, 2020-08-06:
@cdb0y511 hi,
>I figure out how to read the confidence.
It is raw 8 bit value.

Confidence data is **4-bits per pixels** packed so each byte holds the confidence values for two consecutive pixels. 
Please verify that the parsing is consistent with the format's spec- 
https://static.lwn.net/kerneldoc/media/uapi/v4l/pixfmt-cnf4.html

cdb0y511, 2020-08-06:
> consecutive

Thanks for quick reply,
I have read the document , and I am very confused. Cause If I use :

                cfg.enable_stream(RS2_STREAM_CONFIDENCE, 1024, 768, RS2_FORMAT_RAW8,30);// L515

The writed confidence map is the same with viewer. The viewer is also use RS2_FORMAT_RAW8.
I guess SDK unpack the 4-bit to 8-bit per pixel automatically , right?

ev-mp, 2020-08-06:
@cdb0y511 that's correct - the confidence format is accessible in SDK as RAW8 only and the unpacking from CNF4 into RAW8 is performed internally. Currently the SDK does not expose the packed confidence stream.
As noted, during that transition the actual data is relocated into the 4 MSB to enhance visualization. Thus to get it correctly each value should be divided by 16.

cdb0y511, 2020-08-06:
@ev-mp 
Thanks, could you tell me how the confidence calculate？ What is the physical means？
I have done some tested. I guess it is the recived power of each pixel. It means how much photon recived. Higher means higher confidence.
I don't find the infared sensor type in datasheet.
Why dose it need a infared sensor ?
I think it utilize the mems laser scan and apd recive the photon to calculate phase. 
A infared sensor to calculate the recived power? For confidence?

ev-mp, 2020-08-06:
@cdb0y511 , the recently released [L515 User guide](https://support.intelrealsense.com/hc/en-us/articles/360051646094-Intel-RealSense-LiDAR-Camera-L515-User-Guide) is a must read that imo will address most of those inquiries.

Feel free to ask if you need it further elaborated.

cdb0y511, 2020-08-07:
@ev-mp , the user guide is very helpful.
However, I still have some questions. 

" Infrared stream represents the intensity of the reflected laser light off the objects in the scene.
Confidence stream shows the relative confidence level for each computed depth value. "

I think the confidence stream is coincide with infrared stream.
Could you explain the relation between them, some formula will be very helpful.
Cause I find the confidence stream is very useful on filter depth map, custom confidence filter is better than the default 3 level one if I can understand the meaning of confidence thoroughly.
Thanks

TiagoTostas, 2020-11-26:
Is there any example to read the confidence map / filter based on the confidence treshold, using the python wrapper?

I am reading from a .bag, example just adding the important bit: 
 ```
 pipeline = rs.pipeline()

    # Create a config object
    config = rs.config()

    # Tell config that we will use a recorded device from file to be used by the pipeline through playback.
    rs.config.enable_device_from_file(config, path_to_bag)

    # Configure the pipeline to stream the depth stream
    config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
    config.enable_stream(rs.stream.color, 1280, 720, rs.format.rgb8, 30)

    # Start streaming from file
    profile = pipeline.start(config)

    align_to = rs.stream.color
    align = rs.align(align_to)
```

The pipeline runs fine like this, but when I add:

```
    # Configure the pipeline to stream the depth stream
    config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
    config.enable_stream(rs.stream.confidence, 640, 480, rs.format.raw8, 30)
    config.enable_stream(rs.stream.color, 1280, 720, rs.format.rgb8, 30)
```

I get the following:
Traceback (most recent call last):
  File "/home/tro/git/dish_washer/Vision/plate_detection/plate_pointcloud.py", line 65, in <module>
    profile = pipeline.start(config)
RuntimeError: Couldn't resolve requests

Any hints please?

cdb0y511, 2020-11-26:
Hi, @TiagoTostas 
I am not familiar with python SDK.
But I can read it in C++, I hope it can help you.
By the way, what gonna do with the confidence? Any good ideal?
I find the default confidence filter is good for most scenes.

        if(isL515)
            for (auto&& f : frameSetDepth)
        {
            if(f.get_profile().stream_type() == RS2_STREAM_CONFIDENCE){
                frames.confidence_ = std::make_shared<cv::Mat>(Size(width_depth, height_depth), CV_8UC1, (void *) 
                f.get_data(), Mat::AUTO_STEP);
            }
        }

ctlandi, 2020-12-10:
Hi @cdb0y511, I am working in c++ on L515 confidence map too. I enabled the confidence stream but it is not clear to me how to get confidence values online. I think the piece of code you shared can be helpful, but it is not clear to me how you defined frames.confidence_. Would it be possible to share a little more of your code regarding the confidence map part? Thank you

cdb0y511, 2020-12-11:
> Hi @cdb0y511, I am working in c++ on L515 confidence map too. I enabled the confidence stream but it is not clear to me how to get confidence values online. I think the piece of code you shared can [...]

The  frames.confidence_ is just a shared_ptr which points to a cv::mat.

RealSenseSupport, 2021-03-24:
Hi,

Will you be needing further help with this?

*If we don’t hear from you in 7 days, this issue will be closed.*

Thanks

closed 2021-04-01 by cdb0y511

sklipnoty, 2021-06-09:
Can someone please share a minimal example of opening and saving the confidence stream using the python wrapper?

kuvalin, 2021-07-21:
@cdb0y511 I'm also trying to understand how confidence is calculated.  Seems like your questions in [#6775](https://github.com/IntelRealSense/librealsense/issues/6775) answered how to change the threshold, but nowhere in the code is the implementation of RS2_OPTION_CONFIDENCE_THRESHOLD.  The unpacking of the confidence map to RAW8 is done internally, but do you think the confidence calculation is done internally as well?  I've been trying to figure out a way to display the confidence map using librealsense, unfortunately RAW8 is not supported, only greyscale and colour.  Have you been displaying the confidence map using openCV?