realsenseai/librealsense #7030: I have a problem with infrared stream in L515,

helenaBh, 2020-08-04:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

helenaBh, 2020-08-04:
This is my python code:
config = rs.config()
config.enable_stream(rs.stream.infrared, 1, 1024, 768, rs.format.y8, 30)
config.enable_stream(rs.stream.infrared, 2, 1024, 768, rs.format.y8, 30)
config.enable_stream(rs.stream.depth, 1024, 768, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)
# Create a pipeline
pipeline = rs.pipeline()
# Start streaming
profile = pipeline.start(config)

I got this error:
 profile = pipeline.start(config)
RuntimeError: Couldn't resolve requests

helenaBh, 2020-08-05:
Hi, 

I solved the problem by rewriting the code as below:

config.enable_stream(rs.stream.infrared, 1024, 768, rs.format.y8, 30)
config.enable_stream(rs.stream.depth, 1024, 768, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)

When I removed the index for infrared camera the problem solved and I can now capture and save all three infrared, depth, and RGB frames.

closed 2020-08-05 by helenaBh

Sherlock-hh, 2021-07-15:
hello， I have problem with what kind of light the L515 infrared camera can see？I used the D435 infrared to capture 850-860 nm light point , but the L515 can't see this kind light at all. have you try to use L515 capture any kind of light point ? I'll be very appreciate if you have any advice.