realsenseai/librealsense #7033: Problems with file librealsense_mex

lacandelaria, 2020-08-05:
I have read (literally) every issue about this topic, but none helped me out with my problem.

I have been trying to extract depth data from .bag files recorded with the Depth Camera D415 and by using MATLAB functions for a month now, but I got no luck yet. (I have also tried to use python but I couldn't install pyrealsense, neither on a Windows or on a Mac).
This is the first part of the Matlab code I'm using (disclaimer: I have been sent this code by people from my University and it works on their computers):

```
`
pipe = realsense.pipeline();
colorizer = realsense.colorizer();
cfg = realsense.config();
dec_filt = realsense.decimation_filter();
var=[]; 
cm1='C:\path\file_name.bag';
namefile=cm1(end-18:end-4);  
cfg.enable_device_from_file(cm1,0);
profile = pipe.start(cfg);
pointcloud = realsense.pointcloud();
dev=profile.get_device();
out = realsense.librealsense_mex('rs2::device', 'as', dev.objectHandle, 'playback');
plb =realsense.playback(out,-1);
calibrazione=[];
plb.set_real_time(0); %per settare il realtime, non perdo frames
fs = pipe.wait_for_frames();
depth=fs.get_depth_frame();
depthh=depth.get_data();
col=fs.get_color_frame();
data=col.get_data();
align_to = realsense.stream.color;
alignfs=realsense.align(align_to);
aligned_frames = alignfs.process(fs);`
```

and I get always the same error: "Unable to resolve the name realsense.librealsense_mex" (summoned by realsense.pipe (line 10)).  

I'm not pretty sure about what this file is, because I have 7 of them (as you can see in the picture) in a directory called +realsense in the same path as the matlab file I'm using.
[image]
This directory +realsense contains all the file that I downloaded from the GitHub project librealsense/wrappers/matlab/ so I guess that it shouldn't need anything else. I followed the [guide ](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/matlab#building-from-source) of that project, but I couldn't understand how to use CMake and to build the mex file, so maybe that could be a problem. I'm honestly clueless about this whole thing.

I don't know what information I could add to this, I will really appreciate any help you might give.

Peace

MartyG-RealSense, 2020-08-05:
Before looking at why you may be having problems with **librealsense_mex**, it may be worth looking at a simpler solution for extracting depth data from a bag.  The RealSense SDK has a tool called **rs-convert** for extracting frames from a bag file and converting the data into a range of file formats.

https://github.com/IntelRealSense/librealsense/tree/master/tools/convert

[image]

On Windows there is a pre-built version of **rs-convert** in the SDK's **Tools** folder at **C: > Program Files (x86) > Intel RealSense SDK 2.0 > Tools** that you can run from the Windows command prompt interface.

lacandelaria, 2020-08-05:
Thank you very much! I actually didn't know how to use it, but for some reasons I hadn't looked for info about that exe. Myyy bad.

I'd like to ask you something more: is there any way to modify rs.convert.exe or in general to edit the final name of the .csv files?

MartyG-RealSense, 2020-08-05:
As the SDK's tools are open-source code, you could certainly modify the **rs-convert** program in the link that I provided for your own needs and then use a compiler program such as **Visual Studio** to generate your own custom .exe version of the tool.

I believe though that you can set a custom name of the csv output file just by defining it as part of the save path in the extraction command.  For example, in the rs-convert launch command taken from the link below:

rs-convert.exe -v C:\Users\efoo1\Desktop\**test.csv** -i C:\Users\efoo1\Documents\20181105_102448.bag

https://github.com/IntelRealSense/librealsense/issues/2671#issuecomment-435773984

lacandelaria, 2020-08-06:
Again, thank you so much. You helped me a lot!

closed 2020-08-06 by lacandelaria

ph-code-repo, 2021-07-12:
I'm still having the same problem as the original post.
I'm using Linux, so not completely expecting it to work, but found it interesting that I was getting the same compiler problem with the librealsense_mex.mexa64 file not being found despite everything it needs to be created being available.
There appears to be some problem with the phony target declaration, though I don't know enough about this to understand what is going on.
[image]

MartyG-RealSense, 2021-07-12:
Hi @ph-code-repo  There are very few information references for your problem, as the RealSense MATLAB wrapper was designed to work with Windows 10 rather than Ubuntu.  There was one article where someone got it working on Ubuntu and shared their method for creating **librealsense_mex.mexa64**.  At the link below, you can find the details under the **December 4, 2018** heading.

https://staff.fnwi.uva.nl/a.visser/activities/IMAV/Labbook2018.html