realsenseai/librealsense #7037: Context instantiation throws a DllNotFoundException

amitelle1337, 2020-08-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i & L515 | 
| Firmware Version                   | 05.12.06.00 & 01.04.01.02 respectively | 
| Operating System & Version |   Win 10  |  
| Platform                                 | PC |
| SDK Version                            | v2.36.0 | 
| Language                            | C# |
| IDE | Visual Studio 2019 |

### Issue Description
I have been using the pyrealsense for a while and it worked great.
Now, I am trying to transfer to C# (for WPF purpose), and I can't manage to build a project that works.

I am trying to get this simple code running:

```C#
using Intel.RealSense;
using System;

namespace Test
{
    class Program
    {
        static void Main(string[] args)
        {
            var pipe = new Pipeline();
            pipe.Start();

            while (true)
            {
                using (var frames = pipe.WaitForFrames())
                using (var depth = frames.DepthFrame)
                {
                    Console.WriteLine("The camera is pointing at an object " +
                        depth.GetDistance(depth.Width / 2, depth.Height / 2) + " meters away\t");

                    Console.SetCursorPosition(0, 0);
                }
            }
        }
    }
}
```

When I try to run the code I am getting the following error:
[image]
For both of my cameras.

I have been looking at issue [#4495](https://github.com/IntelRealSense/librealsense/issues/4495), but it didn't seem to work for me.
I tried everything in the suggestion of this issue for both x64 and x86 dlls, configurations both of my cameras, but without success (getting the same error).

Any idea what causes the problem and how to resolve it?
Thanks in advance.

dorodnic, 2020-08-05:
Hi @amitelle1337 
The main thing to check is that realsense2.dll exists and can be located by the applications. There are several ways to do it, first locate realsense2.dll on your computer. It may also be named realsense2d.dll, in this case rename it to realsense2.dll.
Add the directory of realsense2.dll to %PATH% environment variable (and restart app and visual studio). You can also try copying the file to visual studio project folder or the output folder. 
Note if the file is under win32 or x64 and set your project settings are configured accordingly.

amitelle1337, 2020-08-05:
Both of the solutions worked!
Thank you very much!

closed 2020-08-05 by amitelle1337

juamar, 2022-02-08:
I'm facing this same issue but with Xamaring, building an Android Application.

[DOTNET] System.TypeInitializationException: The type initializer for 'Intel.RealSense.Context' threw an exception.
[DOTNET] ---> System.DllNotFoundException: realsense2
[DOTNET] at Intel.RealSense.Context..cctor()
[DOTNET] --- End of inner exception stack trace ---
[DOTNET] at mauiExample.Fragments.RealsenseFragment.OnViewCreated(View view, Bundle savedInstanceState) in C:\Users\juan.avendano\Codes\Thales\Singapur\mauiExample\AndroidProject\Fragments\RealsenseFragment.cs:line 28
[DOTNET] System.DllNotFoundException: realsense2
[DOTNET] at Intel.RealSense.Context..cctor()

I've sorted out to build a simple demo App using Xamarin Binding with the Android/Java Wrapper. But will be nice to just use the c# wrapper with Xamarin, which will make our life easier in a project we have to develop. We have to integrate some code that is already using the C# wrapper and .NET Core.

I'm adding both references to the Xamarin project:

[image]

Thanks in advance.