realsenseai/librealsense #7038: Announcement: Intel white paper published on L515 multi-camera configuration

RealSenseSupport, 2020-08-05:
Hi everyone,

Intel have published a new RealSense white-paper document on multiple camera configuration for the L515 lidar depth camera.

https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup

Fred3D-tech, 2020-08-10:
very interesting but you should really give more details about this section: "Appendix
The following section will show a simple code how to generate a sequence of GPIO sync between the cameras over Raspberry Pi 4"

RealSenseSupport, 2020-08-10:
Hi @Fred3D-tech Could you please specify what details you are looking for?

closed 2020-08-19 by RealSenseCustomerSupport

RavindraJal, 2020-08-20:
How to make L515 on Arm processors such as Raspberry PI or Nvidia jetson, has this been tested on arm processors with ros2, if the answer is yes, then I will definitely buy this lidar, seems like the perfect sensor for my robot
Thanks

Fred3D-tech, 2020-08-20:
> Hi @Fred3D-tech Could you please specify what details you are looking for?

sorry but the appendix is very very short.. very interesting but very shot. For example, the second picture showing the connections is very and without captions.. we need a more detailed view of the connections with captions or a video. As is it is, your documentation is just barely usable.
Thanks for all your work but think more about your poor clients/users/buyers..

AH50, 2021-02-18:
Hi, after recording a sequence of images from all cameras, what do you use to reconstruct the scene? Is it Open3d? If so, could I ask for the reconstruction configuration?