realsenseai/librealsense #7040: L515 accuracy data

ihsanmutlu, 2020-08-05:
Hi everybody,
I would like to know  depth accuracy data for L515 depth camera. On datasheet, depth accuracy is given as <5mm at 1m, for VGA resolution and reflective surface. My object will be at 3meters and has a non reflective surface. I will use XGA resolution. Can you give me an average for depth accuracy? And another question: Is L515 depth camera best for my situation? I need submilimeter accuracy.

rafaelspring, 2020-08-06:
> Is L515 depth camera best for my situation? I need submilimeter accuracy.

Can you describe your application?

cdb0y511, 2020-08-06:
Hi, @ihsanmutlu 
You must be aware the accuracy is measured  under specific condition : 
1  face a 95% reflective object.
2. no ambient light.
This is the best accuracy L515 can be.

ihsanmutlu, 2020-08-06:
Thanks for input. Based on cdb0y511's comment, L515 isn't for me.

RealSenseSupport, 2020-08-06:
Hi @ihsanmutlu 

Many factors influence data accuracy/precision. Can you provide some details on your exact usage model?

Thanks

RealSenseSupport, 2020-08-14:
Hi,

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

ihsanmutlu, 2020-08-15:
Thanks. It was helpful. I forgot to close it.

closed 2020-08-15 by ihsanmutlu