realsenseai/librealsense #7043: RealSense Viewer crashes when trying to open with D435i connected

Brac24, 2020-08-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Firmware Version                   | 05.12.06.00 | 
| Operating System & Version |   Windows 10 Build 1909  |
| Platform                                 | PC |
| SDK Version                            |  2.36.0.2034  | 
| Language                            |  C++                         | 
| Segment			|  Robot                         | 

### Issue Description
 RealSense viewer crashes when I try opening it when I have the camera connected. If it is not connected the viewer opens fine.  Opening the viewer then connecting the camera does not work either since the application does not detect the camera but it does show on my PC that the camera is set up and ready to use.
I tried the RealSense viewer exe on its own and the same thing happens.

I tried running the capture example in Visual Studio and an image did appear for about a minute. The updating of the camera capture was extremely slow and once I put my hand in front of the camera in close proximity the capture program crashed and I have since not been able to run it again. The application opens and after a few seconds of loading it exits.

I just ran the hello-realsense program in Visual Studio and the first run it displayed that the camera was facing an object 0 meters away. The second time I ran it, it opened and after a few seconds with a blank console the program seemed to crash and it exited. I actually got it to run once saying that it was facing an object 0.54 meters away.
Using the camera has very sporadic results.

I installed the SDK on a different Windows 10 PC and seemed to work a lot better so it must be my PC's configuration. I honestly don't know where to start I feel like it could be anything.

MartyG-RealSense, 2020-08-06:
Hi @Brac24   There is a case with similar characteristics that is currently being explored.  In that case, the camera model involved is SR305.  The RealSense user in that case has tested with the Viewer installed by the Windows SDK binary and the standalone Viewer, and with rs-capture and hello-realsense.  

The version of the Viewer launched with the desktop icon exited as soon as it launched.  The standalone version launched but depth streaming would freeze.  Depth streaming in **rs-capture** and **hello-realsense** would likewise freeze almost immediately.  In that particular case, RGB did not seem to be affected.

It is s long discussion.  The lengthy diagnostic process eventually pointed towards the possibility that the camera had been negatively affected by a Windows update, as has sometimes occurred in the past.  As the camera works for younon a different PC, this may lend weight to that possibility.

The case is in the link below:

https://github.com/IntelRealSense/librealsense/issues/6959

Brac24, 2020-08-07:
Hi @MartyG-RealSense thank you for your response. No I just received the device yesterday and haven't been able to get it to work yet on my laptop. Although I did update my laptop about a month or 2 ago and it actually caused the laptop's camera to malfunction where the camera turns on then shuts of then on again and it just keeps flickering on and off.

Brac24, 2020-08-07:
Well it looks like it is fixed but I had to disable my laptop's integrated camera. Somehow the integrated camera is affecting the use of the D435i. I think for now I can use it like this thank you for your help.

closed 2020-08-07 by Brac24

MartyG-RealSense, 2020-08-07:
Thanks so much for the pdate - I'm  pleased that you found a workaround for your problem.