realsenseai/librealsense #7047: t265 issue on nvidia AGX Xavier

AndreV84, 2020-08-06:
```
/librealsense-2.38.0/build/examples/pose$ ./rs-pose
RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x559602b0b0):
    Frame didn't arrive within 15000
```

csr-kick, 2020-08-06:
Did this run at all before giving you the error? As in, was this working fine, then just stop?

AndreV84, 2020-08-06:
I wouldn' t say that it did run at all;
We have a bunch of realsense devices and they randomly fail.
Using on nvidia Jetson devices with / without adapters, hubs, powered hubs, unpowered hubs, usb-a ports, usb-c ports;

Did It run? how do I know? somewhere in the past some of t265 cameras worked at some configuration that is all I can see. some other t265 seem working on upsquare x86_64 board. Jetson is arch64;

csr-kick, 2020-08-06:
Sorry, just trying to understand the nature of the failure better. My experience has been that rs-pose will run, printing out an x, y, and z offset to the command line, then randomly stop and give me the error you posted above. For example:
`Device Position: 0.048 -0.002 0.034 (meters)))`

AndreV84, 2020-08-07:
issues change when they are approached;
as for now the app will retutn immediately after running
 rs-pose
RealSense error calling rs2_pipeline_start_with_config(pipe:0x55712faee0, config:0x55712e2f00):
  open(...) failed to configure streams. T265 is running!
On some other device; that has 2.35 version it was said tu run.

AndreV84, 2020-08-07:
after reboot it wil return:
```
 rs-pose
RealSense error calling rs2_pipeline_start_with_config(pipe:0x55af755e50, config:0x55af75df00):
    No device connected
```
can rs-pose start in headless mode at all? or necessarily requires conencted display?

AndreV84, 2020-08-07:
```
 rs-enumerate-devices 
 06/08 20:14:29,770 ERROR [548487543952] (tm-boot.h:39) Error booting T265
 06/08 20:14:32,815 ERROR [548487543952] (tm-boot.h:39) Error booting T265
No device detected. Is it plugged in?

```

csr-kick, 2020-08-07:
`rs-pose` does run headless, yes. 

There are a number of issues with the T265 that you might be facing. First, you cant usually boot the computer, in this case the Xavier, with the T265 plugged in. You must boot and then plug it in - or - boot with it plugged in and then use software to powercycle the USB ports. 

Once you do one of those, check `lsusb`. When the T265 first boots, it will appear as 
`ID 03e7:2150 Intel Myriad VPU [Movidius Neural Compute Stick]`
After you run rs-pose it will load the driver firmware and appear as:
`ID 8087:0b37 Intel Corp.`

AndreV84, 2020-08-07:
is building with keys/arguments belopw the correct thing?
` cmake ../ -DFORCE_RSUSB_BACKEND=true -DBUILD_PYTHON_BINDINGS=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_WITH_CUDA=true -DBUILD_GRAPHICAL_EXAMPLES=true`

csr-kick, 2020-08-07:
Looks fine, but I don't know if there are any other steps you would have to follow like shown in this document
[https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md)

AndreV84, 2020-08-07:
we got the unit disabled;
then powered on Xavier;
after it was powered on we attached powered usb-hub with conencted realsense;
what I am getting before runnign anything is 
```
lsusb
```
Bus 002 Device 003: ID 8086:0b3a Intel Corp. 
Bus 002 Device 002: ID 05e3:0626 Genesys Logic, Inc. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 03e7:2150  
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`````

AndreV84, 2020-08-07:
```
 rs-pose
RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x558d842740):
    Frame didn't arrive within 15000
```

AndreV84, 2020-08-07:
attempt2:
XAvier off; usb hub connected; camera disconnected;
Xavier is on
lsusb
```
Bus 002 Device 002: ID 05e3:0626 Genesys Logic, Inc. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```
camera is added
then a device 
Bus 001 Device 003: ID 03e7:2150

appears

AndreV84, 2020-08-07:
```Bus 001 Device 003: ID 03e7:2150
```
that is the error when I run rs-pose

AndreV84, 2020-08-07:
rs-pose
RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x5583f89b40):
    Frame didn't arrive within 15000

AndreV84, 2020-08-07:
when the camera is connected to a booted without camera unit,
via usb--c to usb-a adapter
after attaching it would show
``
 lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 03e7:2150  
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```

AndreV84, 2020-08-07:
TRYING RS-POSE
 rs-pose
RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x55ab4248c0):
    Frame didn't arrive within 15000
after trying rs-pose
lsusb
Bus 002 Device 002: ID 8087:0b37 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

AndreV84, 2020-08-07:
resolved by removing protected cover from device

closed 2020-08-07 by AndreV84