realsenseai/librealsense #7048: pyrealsense2: RuntimeError: Frame didn't arrive within 5000

JLo9, 2020-08-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435 } | 
| Firmware Version                   |05.12.06.00| 
| Operating System & Version |   {Win 10 | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.36.0.2034 }                          | 
| Language                            |  python3.7.5 64bit }                          | 

### Issue Description
I installed the RealSense SDK for the first time today and want to develop in python. Using RealSenseViewer works without problems.

But whatever I do in python, I run into 

`RuntimeError: Frame didn't arrive within 5000`

The code I use is:

`import pyrealsense2 as rs

rs.log_to_file(rs.log_severity.debug, file_path='./log.txt')

device = rs.context().devices[0]
serial_number = device.get_info(rs.camera_info.serial_number)
print(serial_number)

# Configure depth and color streams
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)

# Start streaming
pipeline.start(config)
frames = pipeline.wait_for_frames()
depth = frames.get_depth_frame()`

It reads the serial number correctly and I can find further information about the camera debugging. But reading frames won't work.

I installed the SDK with the installer. Python is running in a virtual environment. I used  pip to install pyrealsense2.

To log of above code is:

` 06/08 16:41:22,027 DEBUG [16436] (win-helpers.cpp:87) res=pSensorManager->GetSensorsByCategory(SENSOR_CATEGORY_ALL, &pSensorCollection) returned: HResult 0x80070490: "Element nicht gefunden."
 06/08 16:41:22,045 DEBUG [16436] (win-helpers.cpp:87) res=pSensorManager->GetSensorsByCategory(SENSOR_CATEGORY_ALL, &pSensorCollection) returned: HResult 0x80070490: "Element nicht gefunden."
 06/08 16:41:22,061 DEBUG [16436] (ds5-factory.cpp:1109) try_fetch_usb_device(...) failed.
 06/08 16:41:22,061 INFO [16436] (context.cpp:365) Found 1 RealSense devices (mask 254)
 06/08 16:41:22,190 DEBUG [16436] (win-helpers.cpp:87) res=pSensorManager->GetSensorsByCategory(SENSOR_CATEGORY_ALL, &pSensorCollection) returned: HResult 0x80070490: "Element nicht gefunden."
 06/08 16:41:22,206 DEBUG [16436] (ds5-factory.cpp:1109) try_fetch_usb_device(...) failed.
 06/08 16:41:22,206 INFO [16436] (context.cpp:365) Found 1 RealSense devices (mask 254)
 06/08 16:41:22,207 DEBUG [16436] (win-helpers.cpp:87) res=pSensorManager->GetSensorsByCategory(SENSOR_CATEGORY_ALL, &pSensorCollection) returned: HResult 0x80070490: "Element nicht gefunden."
 06/08 16:41:22,225 DEBUG [16436] (ds5-factory.cpp:1109) try_fetch_usb_device(...) failed.
 06/08 16:41:22,225 INFO [16436] (context.cpp:365) Found 2 RealSense devices (mask 255)
 06/08 16:41:22,368 DEBUG [16436] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Die angegebene Datenstromnummer war ungültig."
 06/08 16:41:22,368 DEBUG [16436] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Die angegebene Datenstromnummer war ungültig."
 06/08 16:41:22,369 DEBUG [16436] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Die angegebene Datenstromnummer war ungültig."
 06/08 16:41:22,385 DEBUG [16436] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Die angegebene Datenstromnummer war ungültig."
 06/08 16:41:22,385 DEBUG [16436] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Die angegebene Datenstromnummer war ungültig."
 06/08 16:41:22,385 DEBUG [16436] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Die angegebene Datenstromnummer war ungültig."
 06/08 16:41:22,385 DEBUG [16436] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Die angegebene Datenstromnummer war ungültig."
 06/08 16:41:22,396 INFO [16436] (sensor.cpp:1341) Request: Z16 Depth, 
Resolved to: Z16 Depth, 
 06/08 16:41:22,413 DEBUG [16436] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Die angegebene Datenstromnummer war ungültig."
 06/08 16:41:22,414 DEBUG [16436] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Die angegebene Datenstromnummer war ungültig."
 06/08 16:41:22,414 DEBUG [16436] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Die angegebene Datenstromnummer war ungültig."
 06/08 16:41:22,418 DEBUG [16436] (global_timestamp_reader.cpp:141) time_diff_keeper::start: _users_count = 1
 06/08 16:41:27,432 DEBUG [16436] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Die angegebene Datenstromnummer war ungültig."
 06/08 16:41:27,432 DEBUG [16436] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Die angegebene Datenstromnummer war ungültig."
 06/08 16:41:27,433 DEBUG [16436] (win-helpers.cpp:87) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "Die angegebene Datenstromnummer war ungültig."
 06/08 16:41:27,434 DEBUG [16436] (global_timestamp_reader.cpp:152) time_diff_keeper::stop: _users_count = 0
 06/08 16:41:27,434 DEBUG [16436] (global_timestamp_reader.cpp:155) time_diff_keeper::stop: stop object.
 06/08 16:41:27,434 DEBUG [6448] (global_timestamp_reader.cpp:226) Notification: time_diff_keeper polling loop is being shut-down` 

Happy for any help. Thank you.

MartyG-RealSense, 2020-08-06:
Hi @JLo9   Apologies for the delay in responding.   Do you still experience the error if you add the following line to the end of your script please?

 **pipeline.stop()**

JLo9, 2020-08-06:
Hi @MartyG-RealSense ,

thanks for your help. Adding ``pipeline.stop()`` doesn't change the behaviour. The 

    RuntimeError: Frame didn't arrive within 5000

is an exception thrown at ``frames = pipeline.wait_for_frames()``. Therefore, the script wouldn't reach the stop() bit anyways. Catching the exception and running multiple times doesn't help either. 

I have tried up to 50.000ms for the timeout, but frames just do not arrive.

RealSenseSupport, 2020-08-07:
Hi jLo9,
you can run the python example code python-tutorial-1-depth.py that come with the SDK to see if you are able to get the depth stream ,this sample  use while loop to wait for the frame continuously 
in your code, seems you only wait for the frame once (no while or for loop seen)

JLo9, 2020-08-07:
Hi @RealSenseSupport 

This doesn't work either. The line `frames = pipeline.wait_for_frames()` throws an *Exception* which will cause the python-tutorial-1-depth.py to stop executing after the first iteration.

I can catch the exception inside the loop of course, but just get exceptions in a loop. There are no frames arriving.

If I run the **align-depth2color.py** example, the output will be:

    Depth Scale is:  0.0010000000474974513
    Traceback (most recent call last):
      File "align-depth2color.py", line 47, in <module>
        frames = pipeline.wait_for_frames()
    RuntimeError: Frame didn't arrive within 5000

So it gets a connection to the D435 and can read the `depth_scale`, same as in my example it can read the serial number but as soon as it waits for frames, it fails.

Any other ideas? Thanks.

MartyG-RealSense, 2020-09-03:
Case closed due to no further comments received.

closed 2020-09-03 by MartyG-RealSense

TucoFernandes, 2021-04-30:
I'm having the same issue and it looks like it was never solved....

Required Info |  
-- | --
Camera Model | SR300
Firmware Version | 3.21.0.0
Operating System & Version | Win 10
Platform | PC
SDK Version | 2.43.0
Language | python3.7.1 64bit

In my case, I was just trying to write an app in kivy to capture images and save them to a file. But I have the same issue in all of the python examples. However, if I open Intel Realsense Viewer it connects to the camera. 

Also, I have had some issues on Realsense Viewer as well, all when connecting the depth camera stream.
```
types.h:315 - hr returned: HResult 0x800703e3: "The I/O operation has been aborted because of either a thread exit or an application request."

types.h:315 - _reader->ReadSample(...) returned: HResult 0x8007001f: "A device attached to the system is not functioning."

types.h:315 - _source->QueryInterface(__uuidof(IKsTopologyInfo), reinterpret_cast<void **>(&ks_topology_info)) returned: HResult 0x80004002: "No such interface supported"
sensor.cpp:526 - acquire_power failed: _source->QueryInterface(__uuidof(IKsTopologyInfo), reinterpret_cast<void **>(&ks_topology_info)) returned: HResult 0x80004002: "No such interface supported"

Invalid Value in rs2_open_multiple(sensor:0000021F8E176550,
profiles:0000021F8C9F8558, count:1):

Failed to resolve the request:
	Format: Z16, width: 640, height: 480

Into:
	Formats:
	 INZI
```

MartyG-RealSense, 2021-04-30:
Hi @JLo9   The message about the user holding onto frames after the stream has been stopped is harmless and can be ignored.

In regard to your log warnings about the depth stream when using your SR300 with the Viewer, could you try moving the **Laser Power** setting's slider to the minimal '0' position and then enable the depth stream and see whether the log errors still occur, please?  If they do not then this may indicate that it is a power-related problem, such as the camera dosconnecting because the USB port on the computer cannot supply the camera with its power requirements at that particular moment.

[image]

In regard to the **Invalid value in rs2_open_multiple** problem, could you click on the 'i' information button in the Viewer side-panel and check the **Serial Number** that is displayed for your SR300?  Do they look correct, or do they have a clearly incorrect value such as 'ffffffffffff'?

[image]

TucoFernandes, 2021-04-30:
Hi @MartyG-RealSense, thanks for the quick reply! 
When I set the power to 0 there is no more issue. But 0 means the depth sensor is off right? However, if I put it to only 1 I still get issues with the depth sensor. 

Regarding the **Invalid value in rs2_open_multiple** there is a serial number showing...

[image]

To me, it looks like sometimes the depth sensor connects fine sometimes it doesn't. Because I could connect it with different laser power configurations ...
To tell you the truth I was using this old camera just for starting creating the code since I will be using the D435 in this project. Are there any hardware recommendations? On the website, it just says USB-C 3.1 but I'm afraid now.

MartyG-RealSense, 2021-04-30:
The SR300 is known to have issues related to stability of power supplied to the camera by the USB port on some computers.  Minimizing the Laser Power to '0' and then increasing the laser power in small increments until the camera disconnects is a good test of whether a problem is indeed power-related.  

The 400 Series cameras such as D435 are much more flexible in regard to the computer hardware that they work with, and they can also be used in a USB 2.0 port (though with limited capabilities compared to when they are used with USB 3).  

Details of the 400 Series' support for USB 2 are in the link below.

https://www.intelrealsense.com/usb2-support-for-intel-realsense-technology/
 
In regard to hardware recommendations, any Intel or Arm processor should be able to work with a 400 Series camera.  They also work well on low-power computing devices such as Raspberry Pi.  This is because the 400 Series contains a dedicated processing circuit board called **Vision Processor D4** that performs processing work on camera data at the point of capture so that the computer's CPU and graphics GPU do not have to do all of the processing work.

TucoFernandes, 2021-04-30:
Hi @MartyG-RealSense, just as a quick update, I did plug the camera into another pc (an older win 10 with intel realsense SDK 2.29.0 and there was no issue initially regarding power... I can't say for sure what t solved this particular issue: if the SDK or hardware (although the USB port was 3.0 on both). But for now, I will wait for the D435 before digging more into this. Thanks!

MartyG-RealSense, 2021-04-30:
You are very welcome @TucoFernandes - thanks for the update!

MartyG-RealSense, 2022-03-14:
Hi @carrinodue  The official script pages for the Python examples do not include '5000' in the brackets of the wait_for_frames instruction.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-tutorial-1-depth.py#L25

Do you still experience the error if you remove the number from the brackets and just use **frames = pipeline.wait_for_frames()** please?

carrinodue, 2022-03-15:
Hi @MartyG-RealSense, thank you for your prompt reply. 

If I understand well, ``pipeline.wait_for_frames()`` has 5000 as a default value for the optional parameter *timeout_ms*.
  https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.pipeline.html#pyrealsense2.pipeline.wait_for_frames

But, yes, I tried with and without 5000, with 50'000 and also tried the function ``try_wait_for_frames()``.  While in debug mode it works fine, when I run the script (in PyCharm or in a Jupyter lab) I never get a frame. Thanks

MartyG-RealSense, 2022-03-15:
Yes, the pipeline will timeout after 5 seconds if frames are not received.

It's no trouble at all, it's great to hear that you found a solution for the PyCharm aspect of your problem.  Thanks so much for sharing the method that you found for dealing with this problem  :)

When librealsense is built in **Release** mode then it takes advantage of compiler optimizations.  I have not seen a previous case of Release mode causing problems with a Jupyter project though.

carrinodue, 2022-03-15:
Ok, **final update** from my side. What caused problems to run the code on **Jupyter** was.... Kaspersky endpoint security 😣. I had just to turn it off and everything worked fine. 
Thanks again.

MartyG-RealSense, 2022-03-15:
I'm pleased that you found a solution to the remaining Jupyter issue!