realsenseai/librealsense #7049: Measurement amount window is off screen, cant see amount

rasamaya, 2020-08-06:
| | |
|---|---|
|**librealsense**|2.36.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense L515|
|**Serial Number**|00000000f0211525|
|**Firmware Version**|01.04.01.02|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|00000003a9681eec|
|**Firmware Update Id**|00000003a9681eec|

Please provide a description of the problem

RealSenseSupport, 2020-08-06:
Hi @rasamaya 

Can you provide a more detailed description of the problem?

Thanks

RealSenseSupport, 2020-08-14:
Hi,

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

rasamaya, 2020-09-06:
Description was in the title, and I didnt see this. I get 1k github notes a day. I should follow a little less. The measurement tool ui that shows the measurement amount, is offset, and cannot be seen on the screen most of the time for me. Maybe a resolution scaling thing on my surface??