realsenseai/librealsense #7050: If motion Module toggled, RGB overlay doesnt work anymore

rasamaya, 2020-08-06:
| | |
|---|---|
|**librealsense**|2.36.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense L515|
|**Serial Number**|00000000f0211525|
|**Firmware Version**|01.04.01.02|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|00000003a9681eec|
|**Firmware Update Id**|00000003a9681eec|

Please provide a description of the problem

ev-mp, 2020-08-06:
@rasamaya, thanks for the report. This is a recognized issue with the `viewer` that will be addressed in next releases.
In the meanwhile you can use a workaround to troubleshoot the glitch - https://github.com/IntelRealSense/librealsense/issues/6992#issuecomment-667633901
Duplicate of #6992

RealSenseSupport, 2020-08-11:
Hi @rasamaya,,

Did that work for you? Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks