realsenseai/librealsense #7051: How to contribute a "use case", "press release", or "blog post" for intelrealsense.com

daniel-packard, 2020-08-06:
Hi - this might not be the right forum for this question. However, I suspect other developers might be in the same boat.

I am working for a company that uses Intel RealSense for an exciting healthcare application, and we'd like to share that with the community, e.g. as in this post: https://www.intelrealsense.com/smart-wheelchairs-with-luci/

We have been unable to find the right contacts, or receive followup to coordinate this. 

Can anyone here help us find the way? 

Thanks!

daniel-packard, 2020-08-06:
related: #7044

MartyG-RealSense, 2020-08-06:
Hi @daniel-packard   I do not know of a direct contact for the https://www.intelrealsense.com/ website that the RealSense blog, use cases, etc are featured on.  However, if you do a Twitter post about your product and include the hashtag **#IntelRealSense** then it has a good chance of being featured on the official RealSense Twitter feed.  That should get the attention of Intel RealSense people who are responsible for creating site content.

https://twitter.com/IntelRealSense

daniel-packard, 2020-08-06:
Thanks for the reply @MartyG-RealSense - We tried that route.. with limited success

We connected first via tweet:
[image]

And later by direct message: 
[image]

But we stopped receiving responses

MartyG-RealSense, 2020-08-06:
Since the IntelRealSense website is oriented around information and sales rather than technical support, the best way to make an official first contact with sales & marketing may be to go through the **Ready to talk to sales?  Click here** button at the bottom of the https://www.intelrealsense.com/ page and fill in the menu-driven contact form.  That will give you a starting point for a conversation with the RealSense people in that department.

[image]

daniel-packard, 2020-08-06:
Ok, will give it a shot - thanks for the tips!