realsenseai/librealsense #7052: Install version 2.36 for realsense_ros to work

Joechencc, 2020-08-06:
I need to use realsense camera in ROS, I need to install the software in version 2.36. However, I followed with the guide in https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md. It can only install version 2.35.2 

How can I add 2.36.0 version for Ubuntu 18.04 Amd64 core ?

Joechencc, 2020-08-06:
Simply download the source zip and make from the file will not fully work. When I call roslaunch realsense2_camera rs_camera.launch, [ WARN] [1596735313.450369309]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
 06/08 13:35:13,697 WARNING [547648598400] (ds5-timestamp.cpp:73) UVC metadata payloads not available. Please refer to the installation chapter for details.

And I am using two camera at the same time, D435 and T245. Only D435 works, T245 can be found, but error pops up. Can you share the link that I can uninstall 2.35.2 and install 2.36.0??

ev-mp, 2020-08-06:
@Joechencc hello, 
1)
> However, I followed with the guide in https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md. It can only automatically install version 2.35.2

The version 2.36.0 was uploaded a month ago and is available for Bionic (build 3169).
You should be able to see the packages listed
```sh
$apt-cache showpkg librealsense2 | grep 2.36.0
2.36.0-0~realsense0.3169 (/var/lib/apt/lists/realsense-hw-public.s3.amazonaws.com_Debian_apt-repo_dists_bionic_main_binary-amd64_Packages) (/var/lib/dpkg/status)
  librealsense2-dbg,librealsense2 2.36.0-0~realsense0.3169
  librealsense2-utils,librealsense2 2.36.0-0~realsense0.3169
  librealsense2-net,librealsense2 2.36.0-0~realsense0.3169
  librealsense2-gl,librealsense2 2.36.0-0~realsense0.3169
  librealsense2-dev,librealsense2 2.36.0-0~realsense0.3169
```
In case the command can locale v2.35.2 only it would indicate a connectivity or cache-related issue with the host.
Try to purge and regenerate `apt-cache` and also verify that the PPA is properly registered and `sudo apt update` completes successfully.

2)
>Can you share the link that I can uninstall 2.35.2 and install 2.36.0??

You can use the original link attached to reinstall the packages. It should be similar to
```sh
sudo apt purge librealsense2 -y
sudo apt update
sudo apt install librealsense2-utils librealsense2-*-dev librealsense2-*-dbg
```

3)
>06/08 13:35:13,697 WARNING [547648598400] (ds5-timestamp.cpp:73) UVC metadata payloads not available. Please refer to the installation chapter for details.

Follow the instructions on the guide you linked to patch the kernel: 
- `sudo apt install librealsense2-dkms` for LTS kernels up to 5.3. 
- For kernel 5.4 use the script and patches from `development` branch - https://github.com/IntelRealSense/librealsense/blob/development/scripts/patch-realsense-ubuntu-lts.sh

Joechencc, 2020-08-06:
I was using arch kernel like Jetson NX computer, will this matter? I try to install this manually. The error stays.

MartyG-RealSense, 2020-08-09:
Hi @Joechencc    If you are having difficulty with installing librealsense and ROS separately, I wonder if the **Method 1** ROS installation process may be suitable for you, as it installs librealsense and the ROS wrapper **together**.  It may not install the very latest version of librealsense, but if you do not mind using librealsense version **2.36.0** then it may be suitable for you.

https://github.com/IntelRealSense/realsense-ros#method-1-the-ros-distribution

closed 2020-09-09 by MartyG-RealSense