realsenseai/librealsense #7053: l515 RS camera android app

THOROFHEAVEN, 2020-08-06:
Hello,
I'm trying to connect my L515  to android Huawei p30 pro for first run on RS Camera app, but there is "connect realsense camera" all the time. OTG menager shows device but i cant turn it on. Any ideas?

dorodnic, 2020-08-09:
Are you referring to RS Camera App from Google Play-Store or built from source?
The Play-store App is not yet updated with L500 support, since there are still some known issues (primarily on RK3399) that are being addressed.

THOROFHEAVEN, 2020-08-09:
Thank you for info. I'm trying to test it on adroid with java example with 2.36.0 build and still i cant connect my l515 to work with it. 

-

dorodnic, 2020-08-10:
Please make sure device is connected as USB3. At the moment, when connected as USB2 (either due to cable, connection, or port) the device will show up in the system but will not work. 
What version of Android are you using?

THOROFHEAVEN, 2020-08-10:
android 10.1.0
Now after starting for example capture app on my phone "connnect realsense camera" after l515 connection dissappear and there is black screen. I dont know what is the issue, i have wire from l515 and usb 3.1 otg adapter.

THOROFHEAVEN, 2020-08-12:
Im using Android 10 on Huawei p30 pro.
Device is connected to android with usb 3.2 (app checked) but all
examples showing black screen after l515 connected. I don't know what to do
now.

THOROFHEAVEN, 2020-08-25:
> Please make sure device is connected as USB3. At the moment, when connected as USB2 (either due to cable, connection, or port) the device will show up in the system but will not work.
> What version of Android are you using?

Now after checking log file on my Phone after l515 attach i get dl.getDeviceCount =0.

RealSenseSupport, 2020-10-23:
Hi,

Are you still experiencing these issues with the latest Android app and L515 F/W.

Thanks

THOROFHEAVEN, 2020-10-23:
No,  i have four usb 3.0 connectror and olny one is working with 3. standard now, so its good. In addition I checked deep resolution for incoming frames and there was bad value. Now everything working fine.

RealSenseSupport, 2020-10-23:
That's great news. We'll close this issue then and if you have any further problems/questions feel free to open a new one.

closed 2020-10-23 by RealSenseSupport