realsenseai/librealsense #7054: RS D415 Android(Java) AppyFilter

YichaoW, 2020-08-06:
When I try to do ```applyFilter``` on the frameset, ```df.getDataSize()``` returns 0 and ```vf.getDataSize()``` sometimes returns normal value but other time also returns 0. When I remove ```try (FrameSet alignedFrames = frames.applyFilter(align)) ```, both ```vf``` and ```df``` works normally. Can someone tell me how to use ```applyFilter``` appropriately?

```java

            try (FrameSet frames = pipe.waitForFrames()) {
                Align align = new Align(StreamType.COLOR);
                try (FrameSet alignedFrames = frames.applyFilter(align)) {
                    try (Frame f = alignedFrames.first(StreamType.COLOR, StreamFormat.BGR8)) {
                        VideoFrame vf = f.as(Extension.VIDEO_FRAME);
                    } catch (Exception e) {
                        Log.d(TAG, "video frame error: " + e);
                    }
                    try (Frame fDepth = alignedFrames.first(StreamType.DEPTH, StreamFormat.Z16)) {
                        DepthFrame df = fDepth.as(Extension.DEPTH_FRAME);
                    } catch (Exception e) {
                        Log.d(TAG, "depth frame error: " + e);
                    }
                } catch (Exception e) {
                    Log.d(TAG, "applyfilter error: " + e);
                }

```

liuluyangdev, 2021-07-10:
Hi,bro.How to force frame to Mat(OpenCV)?Thanks

oddityyyy, 2022-02-28:
> When I try to do `applyFilter` on the frameset, `df.getDataSize()` returns 0 and `vf.getDataSize()` sometimes returns normal value but other time also returns 0. When I remove `try (FrameSet [...]

After using applyFilter (), you should apply relseaseWith(fr).

closed 2022-03-04 by YichaoW