realsenseai/librealsense #7055: R200 and odroid N2+ (kernel 4.9.x uvcvideo patch requirement)

fxprime, 2020-08-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200  } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   { Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.9.230-74)                                         | 
| Platform                                 | Odroid N2+ |
| SDK Version                            |  { legacy   }                          | 
| Language                            |  {C }                          | 
| Segment			|  {Robot }                          | 

### Issue Description

Hi there,

I'm working on robot that use Odroid N2+ installed ubuntu mate 18.04 and kernel version is 4.9.x.
I have tried to success patch uvcvideo and it can do 'sudo modprobe uvcvideo' just fine. But when i plug in r200 it turn out freeze and reboot itself. 

As far as i know, currently Odroid N2+ team have no plan to work with kernel 4.15, 4.18, 5.0 or 5.3. 
So i dont have any choice but open issue for help. Any suggestion? If someone have some patch script that can work with kernel 4.9 i will happy to try. Sorry for bad English.
 
Thank you advanced!

MartyG-RealSense, 2020-08-07:
In the modern day RealSense SDK, if there are complications with patching (for example, with a particular computing device) then you can build with the **libuvc backend** method to avoid the need for kernel patching when building librealsense.  Legacy Librealsense had a similar libuvc backend process to bypass the need for patching.

https://github.com/IntelRealSense/librealsense/issues/73

fxprime, 2020-08-07:
> Hi @fxprime In the modern day RealSense SDK, if there are complications with patching (for example, with a particular computing device) then you can build with the **libuvc backend** method to avoid the need for kernel patching when building librealsense. Legacy Librealsense had a similar libuvc backend process to bypass the need for patching.
> 
> #73

Thank you for replied krub. I will try and share the result soon.

fxprime, 2020-08-07:
Update : 

After i try librealsense from

https://github.com/Maghoumi/librealsense

I have some error as show below

```
src/image.cpp:12:14: fatal error: tmmintrin.h: No such file or directory
     #include <tmmintrin.h> // For SSE3 intrinsics used in unpack_yuy2_sse
```

Am i missing something? By the way, i try to force cmakelist (of librealsense version 1.12.4 ) to use libuvc by set 

```
set(BACKEND RS_USE_LIBUVC_BACKEND)
```

After that, i build librealsense 1.12.4 and try to run command

```
cpp-tutorial-1-depth
```
 
It's seem work properly. But when i try to run

```
cpp-config-ui
```

and press "Start capture" The error show up as below

[image]

And terminal error show as below

```
odroid@odroid:~$ cpp-config-ui 
Capturing DEPTH at 628 x 468, fov = 56.6 x 43.8, distortion = NONE
Capturing COLOR at 640 x 480, fov = 54.0 x 41.5, distortion = MODIFIED_BROWN_CONRADY
Capturing INFRARED at 640 x 480, fov = 57.5 x 44.8, distortion = NONE
Capturing INFRARED2 at 640 x 480, fov = 57.5 x 44.8, distortion = NONE
RealSense error calling rs_start_source(device:0x557310ee40, source:VIDEO):
    uvc_set_ctrl(...) returned LIBUSB_ERROR_PIPE
X Error of failed request:  GLXBadDrawable
  Major opcode of failed request:  154 (GLX)
  Minor opcode of failed request:  11 (X_GLXSwapBuffers)
  Serial number of failed request:  15800
  Current serial number in output stream:  15800

```

Any suggestion? 
Thank you advanced krub:)

MartyG-RealSense, 2020-08-07:
Considering that the newest parts of the Maghoumi fork of Legacy Librealsense are 4 years old, it may be too out of date to work now.

It is a good sign that the **cpp-tutorial-1-depth** program was able to work for you after building the official Legacy Librealsense and using the libuvc backend method.  

For the information of others reading this, **cpp-config-ui** is a program for Legacy Librealsense that provides a visual configuration interface like the one that the RealSense Viewer provides in librealsense2.

https://idorobotics.com/2020/02/19/optimising-the-realsense-r200-camera-for-outdoor-use/

Some information on the error that you experienced with it can be found in the link below:

https://github.com/IntelRealSense/librealsense/issues/72

fxprime, 2020-08-07:
> Considering that the newest parts of the Maghoumi fork of Legacy Librealsense are 4 years old, it may be too out of date to work now.
> 
> It is a good sign that the **cpp-tutorial-1-depth** program was able to work for you after building the official Legacy Librealsense and using the libuvc backend method.
> 
> For the information of others reading this, **cpp-config-ui** is a program for Legacy Librealsense that provides a visual configuration interface like the one that the RealSense Viewer provides in librealsense2.
> 
> https://idorobotics.com/2020/02/19/optimising-the-realsense-r200-camera-for-outdoor-use/
> 
> Some information on the error that you experienced can be found in the link below:
> 
> #72

I have tried to update r200 firmware to 1.0.72.10 (Is it lastest version ?)  using FWUpdater as you suggestion.
It worked !  But only when i enable auto LR. When i disable auto LR it turn out as below. I think my version is still not up-to-date i guess? 

[image]

MartyG-RealSense, 2020-08-07:
https://github.com/IntelRealSense/librealsense/issues/73 suggests that the libuvc backend installation method was more experimental than it is in the present day with librealsense2, and that occasional unplug / replug of the camera to reset it may be necessary.

fxprime, 2020-08-14:
> #73 suggests that the libuvc backend installation method was more experimental in the era of librealsense1 than it is in the present day with librealsense2, and that occasional unplug / replug of the camera to reset it may be necessary.

Thank you krub. This can close for now.

closed 2020-08-14 by fxprime