realsenseai/librealsense #7058: No device connected runtime error with Python examples

DrBwts, 2020-08-07:
F200, W10, x63, Python 3.7

I am trying to run the [align-depth2color.py example](https://github.com/IntelRealSense/librealsense/blob/development/wrappers/python/examples/align-depth2color.py) but keep getting the following error,

```

  File "D:\Python codes\OpenCV\align-depth2color.py", line 25, in <module>
    profile = pipeline.start(config)

RuntimeError: No device connected
```

However if I just run a straight up OpenCV video stream capture the device works as expected (although annoyngly onlyin depth mode)

```

import cv2

cam = cv2.VideoCapture(0)

cv2.namedWindow("test")

img_counter = 0

while True:
    ret, frame = cam.read(cv2.IMREAD_COLOR)
    if not ret:
        print("failed to grab frame")
        break
    cv2.imshow("test", frame)

    k = cv2.waitKey(1)
    if k%256 == 27:
        # ESC pressed
        print("Escape hit, closing...")
        break
    elif k%256 == 32:
        # SPACE pressed
        img_name = "opencv_frame_{}.png".format(img_counter)
        cv2.imwrite(img_name, frame)
        print("{} written!".format(img_name))
        img_counter += 1

cam.release()

cv2.destroyAllWindows()
```

MartyG-RealSense, 2020-08-07:
Hi @DrBwts   The RealSense SDK 2.0 that **align-depth2color.py** is part of does not support the **F200** camera, unfortunately,  SDK 2.0's predecessor **Legacy Librealsense** is the SDK that should be used with that camera.

https://github.com/IntelRealSense/librealsense/tree/v1.12.1

DrBwts, 2020-08-07:
Thanks Marty