realsenseai/librealsense #7063: Unclear error message when aligning depth and color frame using pyrealsense2

anguyen216, 2020-08-08:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10) | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.35.2.1937 }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I'm trying to align depth frame to color frame (or vice versa) of a recorded bag file.  My approach is to loop through the bag file frame by frame and aligned depth to color each loop and save the result along the way.  However, I keep getting a vague error message
`RuntimeError: Error occured during execution of the processing block! See the log for more info`

I insert the following block of code in two different locations.  One is before the loop and one is in the loop.  However, I get the same error above both time.
```
# set up alignment
align_to = rs.stream.depth
align = rs.align(align_to)
```
**Before the loop code**
```
# SET UP
cfg = rs.config()
cfg.enable_device_from_file(example_file)
pipe = rs.pipeline()
profile = pipe.start(cfg)
colorizer = rs.colorizer()
# set up alignment
align_to = rs.stream.depth
align = rs.align(align_to)

# get video length
frame_counts = get_num_frames(example_file)

depth_imgs = []
rgb_imgs = []
for i in range(frame_counts):
    frames = pipe.wait_for_frames()
    aligned_frames = align.process(frames)
    # get aligned depth and rgb frames
    aligned_rgb = aligned_frames.get_color_frame()
    aligned_depth = colorizer.colorize(aligned_frames.get_depth_frame())
    depth_imgs.append(np.asanyarray(aligned_depth.get_data()))
    rgb_imgs.append(np.asanyarray(aligned_rgb.get_data()))
pipe.stop()
```
_before the loop full error message_
```
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-32-0c131be6876e> in <module>
     16 for i in range(frame_counts):
     17     frames = pipe.wait_for_frames()
---> 18     aligned_frames = align.process(frames)
     19     # get aligned depth and rgb frames
     20     aligned_rgb = aligned_frames.get_color_frame()

RuntimeError: Error occured during execution of the processing block! See the log for more info
```

**Inside the loop code**
```
# SET UP
cfg = rs.config()
cfg.enable_device_from_file(example_file)
pipe = rs.pipeline()
profile = pipe.start(cfg)
colorizer = rs.colorizer()

# get video length
frame_counts = get_num_frames(example_file)

depth_imgs = []
rgb_imgs = []
for i in range(frame_counts):
    frames = pipe.wait_for_frames()
    # set up alignment
    align_to = rs.stream.depth
    align = rs.align(align_to)

    aligned_frames = align.process(frames)
    # get aligned depth and rgb frames
    aligned_rgb = aligned_frames.get_color_frame()
    aligned_depth = colorizer.colorize(aligned_frames.get_depth_frame())
    depth_imgs.append(np.asanyarray(aligned_depth.get_data()))
    rgb_imgs.append(np.asanyarray(aligned_rgb.get_data()))
pipe.stop()
```
_Inside loop full error message_
```
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-33-9acbba093f81> in <module>
     20     # get aligned depth and rgb frames
     21     aligned_rgb = aligned_frames.get_color_frame()
---> 22     aligned_depth = colorizer.colorize(aligned_frames.get_depth_frame())
     23     depth_imgs.append(np.asanyarray(aligned_depth.get_data()))
     24     rgb_imgs.append(np.asanyarray(aligned_rgb.get_data()))

RuntimeError: Error occured during execution of the processing block! See the log for more info
```

Please let me know how to fix this issue as soon as possible.  This is for a project and I have a deadline to catch

Also a side question, does alignment affect depth values? For example, after alignment process, if I perform object detection on the RGB frame to single out object of interest and then apply the detected mask location on the depth frame (so I only process the information within the area of the mask), would the depth values of the detected object (i.e. depth values within the mask) be accurate/the same as if without alignment?

Fred3D-tech, 2020-08-08:
same problem for me with the last firmware and L515.. never seen that before...

MartyG-RealSense, 2020-08-08:
Hi @anguyen216  The **RuntimeError: Error occured during execution of the processing block!** error occurred in a similar case last month where a RealSense user was trying to align frames from a bag using Python.  I have seen other cases like this too.

https://github.com/IntelRealSense/librealsense/issues/6358

I hope that the link below will be a useful reference for you.

https://github.com/IntelRealSense/librealsense/issues/6330

@Fred3D-tech  If the information above does not solve your L515 related problem, can you create a new case specifically for the L515 please, as someone else on the RealSense team other than myself is responsible for supporting L515 cases.  Thanks!

anguyen216, 2020-08-08:
Following the reference solve my issue! Thank you!

closed 2020-08-08 by anguyen216

MartyG-RealSense, 2020-08-08:
Great news @anguyen216  - thanks so much for the update!

RoyaLxPole, 2024-09-18:
You can do a workaround with a deepcopy of the frame before you do post processing