realsenseai/librealsense #7065: rs2::pointcloud calculate method timed out randomly

lipeize-pku, 2020-08-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                        | D435/D435i | 
| Firmware Version                    | 5.12.06.00 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)   |  4.9.140-tegra    | 
| Platform                                   | NVIDIA Jetson TX2 |
| SDK Version                            |  2.35.2                   | 
| Language                                | C++  | 
| Segment			          |  Robot  | 

### Issue Description

I am using gstreamer to produce a video stream of RGB frames from D435/D435i, while some point cloud algorithms running in the background, but the stream keeps getting cut out after running a while. 

Sometimes it occurs after streaming for 5 hours straight, sometimes after running just 10 minutes. The time of stream cutting out tends to be longer if the system starts for the first time. And if stream restarts immediately after cutting out, it would fail again in just minutes.

After debugging, this line in my code just keeps timing out, and seems to be the reason the stream fails:

```
points = pointcloud.calculate(depth_frame);
```

It just stops here, does not produce any error, or stop the application from running, or proceed to the next line. The CPU usage goes down as algorithms stops after receiving no data, but the D435 camera is still busy and heating.

This also did not happen when I was using 2.13.0 version of the SDK, but only happens after I upgraded to 2.35.2.

Run realsense-viewer after quitting the stream, several errors occured:

```
USB CAM overflow
USB REC overflow
Error polling loop is not behaving as expected!
This can indicate an issue with camera firmware or the underlying OS…
```

I know this issue is super strange. I've never seen it before. It would help a lot if anyone can share some information on what is potentially the problem, or how to debug this.

MartyG-RealSense, 2020-08-08:
Hi @lipeize-pku  If you were able to run the stream with a bag file on loop with gstreamer for days then that suggests that it is not the gstreamer aspect of the project that has the problem.  Nevertheless, you may be interested in the discussion about gstreamer streaming with RealSense in the link below if you have not seen it already:

https://github.com/IntelRealSense/librealsense/issues/6465

There is a known **memory leak** in the point cloud processing block that has been around for a while.  Its first recorded entry in the Known Issues section of the SDK release notes is actually in the **2.18.0** SDK version, and so began after the time at which you were successfully using the program with SDK 2.13.0.

A memory leak is where memory is not released properly by an application and so the memory capacity gets progressively more consumed, until stability of the program's execution degrades and the program may end up freezing or crashing.  

The link below offers advice on how to workaround the point-cloud processing block's memory leak issue:

https://github.com/IntelRealSense/librealsense/issues/2860

lipeize-pku, 2020-08-09:
Thanks for your reply!

I followed instructions in #2860 and put the creation of `rs2::pointcloud pc` variable outside the main data loop, and it solves my failure at line `points = pc.calculate(depth_frame);`

However, another failure arises. This time after running a much longer time. The `pipe.wait_for_frames()` line would report `frame didn't arrive within 15000`, after running several hours.

I tried the method described in #5713, and reset the camera after running some time, with the following code:

```
rs2::context ctx;
rs2::device dev = ctx.query_devices().front();
void reset_camera() {
        dev.hardware_reset();
        rs2::device_hub hub(ctx);
        dev = hub.wait_for_device();
        pipe.start(cfg);
}
```

The application reports error `failed to set power state` when I call `reset_camera()`.

MartyG-RealSense, 2020-08-09:
Could you test whether the situation improves if you move ** pipe.start(cfg);** outside of the **reset_camera()** void statement, please?

lipeize-pku, 2020-08-09:
There is only one camera.

lipeize-pku, 2020-08-09:
Thanks! I'm trying the moving out `pipe.start(cfg)`method.

But since when I debugged, the application stops at `dev.hardware_reset()` and did not proceed any further, I doubt it helps.

At first the function has the first 2 lines in it:
```
void reset_camera() {
        rs2::context ctx;
        rs2::device dev = ctx.query_devices().front();
        dev.hardware_reset();
        rs2::device_hub hub(ctx);
        dev = hub.wait_for_device();
        pipe.start(cfg);
}
```
Then the application would crash at `rs2::device dev = ctx.query_devices().front()` with the same `failed to set power state`. 

It's really strange.

MartyG-RealSense, 2020-08-09:
My understanding is that you cannot use pipe.stop() before pipe.start(), which may explain your crash when doing so.

Usually, hardware_reset() is not used inside a void function but instead is a part of the main block of setup code before the pipe.start() line.  So I am not sure what effect putting it inside a void may have (e.g duplicating configuration functions that may already have been defined in the main block of startup code and should only be used once).

The approach that some users have taken is to program in a hardware_reset() instruction into their application that activates on satisfaction of a certain condition.  For example, if a certain percentage of the depth pixels (e.g greater than 50%) have a value of zero (indicating that something has gone wrong with the image) then initiate a camera reset.

lipeize-pku, 2020-08-12:
Why would the camera suddenly "frame didn't arrive within 15000" after running hours tho? It happens after 5 hours at 30 fps, or 30 hours at 6-7 fps (algorithm slowing down frame rate).

MartyG-RealSense, 2020-08-12:
@lipeize-pku   If the time until an application becomes unstable depends upon the FPS that is being used (with failure occurring faster at higher FPS and taking longer at lower FPS), this can be a symptom of a memory leak.  The higher the FPS, the faster that the computer's memory resources are consumed.

lipeize-pku, 2020-08-12:
I observe that the RAM usage of my system stays around 3G/8G, even when the camera stops, so it's far from filling up. Can it be some other internal resources depletion within the camera?

MartyG-RealSense, 2020-08-12:
What are the camera ASIC and projector temperatures like during your long-run tests, please?  If the projector temperature exceeds 60 degrees C for several seconds during operation then a laser safety mechanism halves the camera's laser power.   If the temperature continues to exceed 60 degrees then the laser is shut off.

lipeize-pku, 2020-08-13:
So I'm testing for temperature in long run in realsense-viewer, with laser power turned to max and a cooling fan keeping the camera at 35℃. Then I noticed some wired behaviour:

After running a while, the frame would stuck every few seconds, FPS drops from 30 to 4, and Output console says: "Out of frame resources!" and "Exception was thrown during user processing callback!"

I suspect it could link to my "frame didn't arrive within 15000" within my own application. 

But I seem to only find "Out of frame resources!" issues in librealsense-ros git. Any idea why it would happen in Ubuntu 18.04 and librealsense 2.35.2?

MartyG-RealSense, 2020-08-13:
It is a good idea to have a cooling fan to help reduce temperature by keeping the outer casing cool.  Be careful not to blow air directly into the camera's ventilation grille though, as pushing air inside can disrupt the airflow inside the device that normally helps warm air to be extracted.

35 degrees C is a fairly normal result for max laser power use over a prolonged period.  I would not expect to see problems starting to manifest in the camera unless the temperature exceeded 42 degrees.

Next, I would suggest monitoring your computer's CPU usage during running and see what it is doing when the FPS drops occur.   If it is "maxing out" at 100% CPU usage then this may create processing bottlenecks so that the CPU cannot cope with the amount of frames that are arriving (hence "out of frame resources" because it is not able to process them in a sufficiently timely manner).

lipeize-pku, 2020-08-26:
I'm closing this issue because the discussion is now around another issue. I'm starting a new issue to address this.

closed 2020-08-26 by lipeize-pku

MartyG-RealSense, 2020-08-26:
Okay, thanks very much for the update.